raspberrypi/pico-sdk #2375: picotool version check uses system path even when PICOTOOL_FETCH_FROM_GIT_PATH set

jondurrant, 2025-03-27:
I have two version of picotool available in my ~/pico directory: picotool-2.1.0 and picotool-2.1.1. Setting environment variable PICOTOOL_FETCH_FROM_GIT_PATH according to documentation should set which picotool to use but this always checks the version in /usr/local/bin/ and not the path pointed to by PICOTOOL_FETCH_FROM_GIT_PATH.

lurch, 2025-03-27:
Could you please clarify the exact paths to your picotool executables, and what you've been setting your `PICOTOOL_FETCH_FROM_GIT_PATH` variable to?

will-v-pi, 2025-03-27:
Also, just to clarify, `PICOTOOL_FETCH_FROM_GIT_PATH` should be set to an empty directory which the SDK will then clone and build `picotool` in - it should not be set to an existing picotool clone that you’ve created (which would probably cause the issue shown here)

jondurrant, 2025-03-27:
Yes the directory isn't empty it is the project folder for the version 2.1.1 of Picotool.
PICOTOOL_FETCH_FROM_GIT_PATH=/Users/jondurrant/pico/picotool-2.1.1

In previous SDK version this was the approach to providing a consistent copy of picotool that can be used across projects. How do we now configure the project not to rebuild picotool?

lurch, 2025-03-27:
> How do we now configure the project not to rebuild picotool?

https://github.com/raspberrypi/picotool?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk

will-v-pi, 2025-04-14:
> Yes the directory isn't empty it is the project folder for the version 2.1.1 of Picotool.

Just to conclude this issue - this is what's causing the problem. If you want to use `PICOTOOL_FETCH_FROM_GIT_PATH` then you have to point it at an initially empty directory that the SDK will populate. You can then point this and any other projects at this using the same `PICOTOOL_FETCH_FROM_GIT_PATH`.

If you want to force fetching from git to ensure it never uses the system path, then you can set `PICOTOOL_FORCE_FETCH_FROM_GIT`

However, the recommended approach is to build and install picotool separately, as described in the [picotool readme](https://github.com/raspberrypi/picotool?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk)