raspberrypi/pico-sdk #2376: Pico 2's core1 halted due to debug-request, after multiple dormant - wakeup cycles

uugear, 2025-03-27:
### Description
I have observed that on Pico 2, after multiple dormant mode cycles, the RP2350 may eventually enter a halted state during one of the dormant transitions.

To reproduce this issue, I use a modified powman example (the original version is [here](https://github.com/peterharperuk/pico-examples/blob/35db8fc31d322de2bcac641d9b6b0f6fdb73062c/powman/powman_example.c), and the reference version I modified is [here](https://github.com/raspberrypi/pico-examples/issues/530#issuecomment-2330842210)):
My modified version: [powman_standalone.zip](https://github.com/user-attachments/files/19485003/powman_standalone.zip)

This example performs a simple task:
- It turns on the LED for 50ms.
- It then puts the RP2350 into dormant mode.
- After 450ms, the RP2350 wakes up.

The expected behavior is that the LED blinks every 500ms indefinitely.
However, in practice, the LED stops blinking after a few cycles (sometimes after dozens of cycles).

I tested whether changing the timing parameters could prevent the issue, but the problem persists regardless of the timing values. Shorter time intervals tend to trigger the issue more quickly.

### Debugging Attempts
I used a Raspberry Pi Debug Probe + OpenOCD for debugging. When the issue occurs, OpenOCD outputs the following logs:

> Debug: 3594 7333 adi_v5_swd.c:404 swd_connect_single(): DP IDR 0x4c013477
Info : 3595 7333 adi_v5_swd.c:498 swd_connect_single(): SWD DPIDR 0x4c013477
Debug: 3596 7333 arm_adi_v5.c:783 dap_dp_init(): rp2350.dap
Debug: 3597 7333 arm_adi_v5.c:814 dap_dp_init(): DAP: wait CDBGPWRUPACK
Debug: 3598 7333 arm_adi_v5.h:682 dap_dp_poll_register(): DAP: poll 4, mask 0x20000000, value 0x20000000
Debug: 3599 7344 arm_adi_v5.c:822 dap_dp_init(): DAP: wait CSYSPWRUPACK
Debug: 3600 7344 arm_adi_v5.h:682 dap_dp_poll_register(): DAP: poll 4, mask 0x80000000, value 0x80000000
Debug: 3601 7344 cmsis_dap.c:845 cmsis_dap_swd_write_from_queue(): refusing to enable sticky overrun detection
Debug: 3602 7345 arm_adi_v5.c:933 mem_ap_init(): MEM_AP CFG: large data 0, long address 0, big-endian 0
Debug: 3603 7345 target.c:2563 target_read_u32(): address: 0xe000ed00, value: 0x411fd210
Info : 3604 7345 cortex_m.c:2547 cortex_m_examine(): [rp2350.dap.core0] Cortex-M33 r1p0 processor detected
Debug: 3605 7345 cortex_m.c:2562 cortex_m_examine(): [rp2350.dap.core0] cpuid: 0x411fd210
Debug: 3606 7345 target.c:2563 target_read_u32(): address: 0xe000ef40, value: 0x10110021
Debug: 3607 7346 target.c:2563 target_read_u32(): address: 0xe000ef44, value: 0x11000011
Debug: 3608 7346 cortex_m.c:2589 cortex_m_examine(): [rp2350.dap.core0] Cortex-M33 floating point feature FPv5_SP found
Debug: 3609 7346 target.c:2563 target_read_u32(): address: 0xe000efb8, value: 0x000000ff
Debug: 3610 7346 target.c:2563 target_read_u32(): address: 0xe000edf0, value: 0x03100000
Debug: 3611 7346 target.c:2651 target_write_u32(): address: 0xe000edf0, value: 0xa05f0001
Debug: 3612 7347 cmsis_dap.c:945 cmsis_dap_swd_read_process(): SWD ack not OK @ 2 WAIT
Debug: 3613 7347 cmsis_dap.c:1257 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 3614 7347 cmsis_dap.c:1272 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 3615 7348 adi_v5_swd.c:404 swd_connect_single(): DP IDR 0x4c013477
Info : 3616 7348 adi_v5_swd.c:498 swd_connect_single(): SWD DPIDR 0x4c013477
Debug: 3617 7348 arm_adi_v5.c:783 dap_dp_init(): rp2350.dap
Debug: 3618 7348 arm_adi_v5.c:814 dap_dp_init(): DAP: wait CDBGPWRUPACK
Debug: 3619 7348 arm_adi_v5.h:682 dap_dp_poll_register(): DAP: poll 4, mask 0x20000000, value 0x20000000
Debug: 3620 7348 arm_adi_v5.c:822 dap_dp_init(): DAP: wait CSYSPWRUPACK
Debug: 3621 7349 arm_adi_v5.h:682 dap_dp_poll_register(): DAP: poll 4, mask 0x80000000, value 0x80000000
Debug: 3622 7350 cmsis_dap.c:845 cmsis_dap_swd_write_from_queue(): refusing to enable sticky overrun detection
Error: 3623 7351 arm_adi_v5.c:583 mem_ap_write(): Failed to write memory at 0xe000edf4
Debug: 3624 7351 target.c:2658 target_write_u32(): failed: -5
Error: 3625 7351 target.c:681 target_examine_one(): [rp2350.dap.core0] Examination failed
Debug: 3626 7351 target.c:682 target_examine_one(): [rp2350.dap.core0] examine() returned error code -5
Debug: 3627 7351 target.c:1775 target_call_event_callbacks(): target event 20 (examine-fail) for core rp2350.dap.core0
User : 3628 7351 target.c:3008 handle_target(): Examination failed, GDB will be halted. Polling again in 700ms
Debug: 3629 7552 cmsis_dap.c:945 cmsis_dap_swd_read_process(): SWD ack not OK @ 3 WAIT
Debug: 3630 7552 target.c:1775 target_call_event_callbacks(): target event 0 (gdb-halt) for core rp2350.dap.core1
User : 3631 7552 target.c:3001 handle_target(): Polling target rp2350.dap.core1 failed, trying to reexamine
Debug: 3632 7552 target.c:675 target_examine_one(): [rp2350.dap.core1] Examination started
Debug: 3633 7552 target.c:1775 target_call_event_callbacks(): target event 19 (examine-start) for core rp2350.dap.core1
Debug: 3634 7552 cmsis_dap.c:1257 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 3635 7552 cmsis_dap.c:1272 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 3636 7552 adi_v5_swd.c:404 swd_connect_single(): DP IDR 0x4c013477
Debug: 3637 7553 cmsis_dap.c:945 cmsis_dap_swd_read_process(): SWD ack not OK @ 1 WAIT
Debug: 3638 7563 cmsis_dap.c:945 cmsis_dap_swd_read_process(): SWD ack not OK @ 1 WAIT
Debug: 3639 7574 cmsis_dap.c:945 cmsis_dap_swd_read_process(): SWD ack not OK @ 1 WAIT
Debug: 3640 7585 cmsis_dap.c:945 cmsis_dap_swd_read_process(): SWD ack not OK @ 1 WAIT
Info : 3641 7595 adi_v5_swd.c:498 swd_connect_single(): SWD DPIDR 0x4c013477
Debug: 3642 7595 arm_adi_v5.c:783 dap_dp_init(): rp2350.dap
Debug: 3643 7595 arm_adi_v5.c:814 dap_dp_init(): DAP: wait CDBGPWRUPACK
Debug: 3644 7595 arm_adi_v5.h:682 dap_dp_poll_register(): DAP: poll 4, mask 0x20000000, value 0x20000000
[... 194 lines cut ...]
Debug: 3838 8170 cortex_m.c:1079 cortex_m_poll_smp(): [rp2350.dap.core1] sending postponed target event 'halted'
Debug: 3839 8170 target.c:1775 target_call_event_callbacks(): target event 0 (gdb-halt) for core rp2350.dap.core1
Debug: 3840 8170 target.c:1775 target_call_event_callbacks(): target event 1 (halted) for core rp2350.dap.core1
User : 3841 8170 armv7m.c:741 armv7m_arch_state(): [rp2350.dap.core1] halted due to debug-request, current mode: Thread
xPSR: 0x09000000 pc: 0x000000da msp: 0xf0000000

### Hardware Affected
This issue is reproducible on my:
- Two Pico 2 boards
- Two custom RP2350-based boards

The affected RP2350 chips (as far as I know) are shown in the following images:
[image] [image] [image]

Interestingly, I have one (and only one) Pico 2 board that does not exhibit this issue. Its RP2350 version matches the one shown in the second image, but another Pico 2 with the same chip version consistently reproduces the problem. They were from the same production batch and I bought them together.

### Further Investigations
I have tried several approaches to resolve or work around the issue, including:
- Resetting core1 at different points.
- Modifying the system clock frequency.
- Adding delays at various locations.

So far, I have not found any reliable solution or workaround.

Any insights into this issue would be greatly appreciated.

peterharperuk, 2025-03-27:
You mention "going dormant" multiple times. Going dormant is something else entirely. "Dormant mode" switches clocks off until woken up. In your case, Powman is actually switching power off to parts of the chip.

uugear, 2025-03-27:
> Powman is actually switching power off to parts of the chip.

Indeed. But it should still wake up (or restart?) the chip when the wake up source is configured.

The example code works as expected until the halting (randomly) happens, and I don't know why the halting would happen.

uugear, 2025-03-29:
I changed the issue title and rephased the text to describe the issue more accurately.

@peterharperuk I am using a modified version of your powman example. I learn from [this discussion](https://github.com/raspberrypi/pico-examples/issues/530) that there will be a non-hardware API to put Pico 2 to low power/DORMANT state, which probably will use the same code from your powman example.

I do think this issue should be addressed: if there is no garantee to wakeup from the DORMANT state, what is the point to have such low power state?

peterharperuk, 2025-03-29:
Yes, I'll take a look. My first guess is that it has woken up but it's crashing somewhere