raspberrypi/pico-sdk #2377: pio_encode_wait_gpio does not respect gpiobase=16

qdotme, 2025-03-27:
Consider the following (equivalent) blocks - one stalling on PIO, one on SIO. Got it to working by manually substracting the 16 from the GPIO pin number (pin_vsync = 45 in my design). 

The PIO is correctly configured for supporting gpio_base=16, compiling the pioasm instructions correctly - but not when generated through `pio_encode_wait_gpio` 

```
#if 1
pio_sm_exec(config->pio, config->pio_sm, 
	pio_encode_wait_gpio(1, config->pin_vsync - 16));
#else
while (gpio_get(config->pin_vsync) == 0);
#endif 
```

kilograham, 2025-03-27:
I guess this is really a documentation issue

Worth looking at any similar functions and see if it is worth adding a variant which takes a PIO instance

kilograham, 2025-07-25:
fix merged into develop; will be in next docs release

PR #2579 (2025-07-25): pico_encode_gpio_wait doc improvement
fixes #2377
closed 2025-07-25 by kilograham