raspberrypi/pico-sdk #2378: How to using  `pico_set_binary_type(freertos_kernel copy_to_ram)`?

yizi15, 2025-03-28:
I only want make some code run in ram. When use   `pico_set_binary_type(freertos_kernel copy_to_ram)`,
Cmake:
```
set(FREERTOS_KERNEL_PATH F:/PICO/pico-sdk/lib/FreeRTOS-Kernel)
if (NOT FREERTOS_KERNEL_PATH AND NOT DEFINED ENV{FREERTOS_KERNEL_PATH})
    message("Skipping FreeRTOS examples as FREERTOS_KERNEL_PATH not defined")
    return()
endif()
list(APPEND PICO_CONFIG_HEADER_FILES ${CMAKE_CURRENT_LIST_DIR}/board_cns.h)

include(FreeRTOS_Kernel_import.cmake)
pico_set_binary_type(freertos_kernel copy_to_ram)
```
and error message
```
[cmake]   set_target_properties Can not find target to add properties to:
[cmake]   freertos_kernel
```

yizi15, 2025-03-28:
Function such as 'memcpy' will be inplace in flash? If i use a forloop, Will it be Optimized to memcpy by complie? I want some function run duration erase flash.

lurch, 2025-03-28:
You might want to use https://github.com/raspberrypi/pico-examples/tree/master?tab=readme-ov-file#freertos as a starting point?

> Function such as 'memcpy' will be inplace in flash?

I believe that if you use copy_to_ram then **al*l* code will be moved to ram.

> I want some function run duration erase flash.

See some of the examples at https://github.com/raspberrypi/pico-examples/tree/master?tab=readme-ov-file#flash

yizi15, 2025-03-28:
@lurch but all my code is too large to place  all in ram now

lurch, 2025-03-28:
In that case I guess you might need to start playing around with linker files, to determine what does or doesn't appear in RAM, but I'm afraid that's not something I personally know anything about.
You could try asking for help on the [forums](https://forums.raspberrypi.com/viewforum.php?f=143), until a more knowledgeable engineer reads this issue?

kilograham, 2025-04-04:
closing as this is a question not an issue, an the solution requires using a custom linker script. see #398 for a placeholder issue for allowing minor changes to exsting linker scripts

closed 2025-04-04 by kilograham