raspberrypi/pico-sdk #2380: More IRQs missing from RP2350.svd

lurch, 2025-03-28:
Similar to #1898 , but slightly different...

I recently noticed that Table 94 in https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf (on page 83) lists IRQ numbers 40 and 41 as `PROC0_IRQ_CTI` and `PROC1_IRQ_CTI`, but these appear to be missing from the `RP2350.svd`.

matiasilva, 2025-07-21:
The CTI allows chip debug. The debug hardware is not bus addressable, so I'm not sure it belongs in the SVD? We never added the RISC-V registers to the SVD, for example.

matiasilva, 2025-07-22:
I'll rename the existing virtual peripheral and add all IRQs there

matiasilva, 2025-07-22:
This is now generating internally and ready for SDK 2.2

kilograham, 2025-07-22:
closing for tracking purposes as it will be in 2.2.0 even though it isn't in develop yet