raspberrypi/pico-sdk #2386: sm_config_set_in_pin_count subtracts one from in_count

visser-jcv, 2025-03-31:
In my opinion, this function should not subtract 1 from in_count.

from pio.h
```
static inline void sm_config_set_in_pin_count(pio_sm_config *c, uint in_count) {
#if PICO_PIO_VERSION == 0
    // can't be changed from 32 on PIO v0
    ((void)c);
    valid_params_if(HARDWARE_PIO, in_count == 32);
#else
    valid_params_if(HARDWARE_PIO, in_count && in_count <= 32);
    c->shiftctrl = (c->shiftctrl & ~PIO_SM0_SHIFTCTRL_IN_COUNT_BITS) |
                   ((in_count - 1) << PIO_SM0_SHIFTCTRL_IN_COUNT_LSB);
#endif
}
```

I have a pio program which calls IN PINS 2 reading input pulses from a motor shaft encoder.
The first pin is used to measure the pulse width. The pulse on the second pin is offset by a +-quarter phase, to indicate possible reverse direction.
So if my pulse measurement is triggered by a rising voltage on pin A - if pin B is high, the motor is reversing, otherwise it's going in a forward direction.

I've got two motors, so I run the same PIO program on two state machines,  for pins 10&11 on one and pins 27&78 for the other
Until now, I never seemed to get the second bit value for either encoder, indicating the reverse direction.
I now call `sm_config_set_in_pin_count(pio, sm, 2+1);` and it works as intended.

visser-jcv, 2025-03-31:
I'm so sorry, I see that it has been changed in latest version of the SDK

closed 2025-03-31 by visser-jcv

visser-jcv, 2025-03-31:
> You might also find https://github.com/raspberrypi/pico-examples/tree/master/pio/quadrature_encoder and https://github.com/raspberrypi/pico-examples/tree/master/pio/quadrature_encoder_substep interesting 🙂
>
>
>
> You might also find https://github.com/raspberrypi/pico-examples/tree/master/pio/quadrature_encoder and https://github.com/raspberrypi/pico-examples/tree/master/pio/quadrature_encoder_substep interesting 🙂

Thanks

 

I will

 

Cheers

Ian

lurch, 2025-03-31:
Just for future cross-reference purposes: this was fixed by #1880

visser-jcv, 2025-03-31:
Andrew – 

 

I looked at the examples you suggested – the JMP mechanism is an interesting one.

If you’re interested in my approach (pulse width resolution of 6 CPU cycles), herewith my PIO source.

I won’t be offended if you bin it 😊

 

Kind regards

Ian

lurch, 2025-04-01:
Hi @visser-jcv , you're not replying directly to me, you're replying to all of GiHub 😉  Also, GiHub automatically strips any email attachments you send to it, so nobody got to see your PIO source.

lurch, 2025-04-01:
> Your example used the rxFifo queue, probably a hangover from RP2040.

Just to clarify - it's not **my** example, it was contributed by @pmarques-dev (see https://github.com/raspberrypi/pico-examples/commits/develop/pio/quadrature_encoder )

pmarques-dev, 2025-04-01:
I think the substep quadrature encoder has several advantages over this code:
- it gives you not only speed, but also proper step counts, i.e., if you're using this to control a CNC machine, for instance, you can keep track of both the position and speed, without losing any steps
- the resolution doesn't degrade at high speeds
- it doesn't use RP2350 specific instructions, so works on RP2040

I've already converted the substep example into a Arduino library (that can be used on Platform.io as well): https://github.com/pmarques-dev/PicoEncoder
The substep code does use the RX queue, but just "on request" (it doesn't flood it) and it guarantees that both values it needs to fetch are an "atomic snapshot", so that you don't read one value before some update and the other value after some update

visser-jcv, 2025-04-01:
fair enough - I just need an accurate speed value for a feedback loop