raspberrypi/pico-sdk #2406: Incorrect results of sin() and cos() for some input values in develop branch?

richard9999999999, 2025-04-09:
I've found that after update to current develop branch I see "glitches" in sin() and cos() results for some input values, e.g:

float angle = 1.56525815f;
float sin_angle = sin(angle);
float cos_angle = cos(angle);

Value of variable sin_angle  is -0.547458887
Value of variable cos_angle is -0.836832583

kilograham, 2025-04-09:
- when did it last work?
- what was the exact answer then?
- are you selecting any variants of `pico_double` implementation?
- what compiler & version are you using

kilograham, 2025-04-09:
also did you recreate your build directory after upgrade

richard9999999999, 2025-04-09:
when did it last work?
>>Today morning :-) when I've used some older SDK based on version 2.1.  Also 2.0 has worked before

> what was the exact answer then?
Hm. Correct values similar to those you await and get e.g. from calculator. Values I am getting (please see previous post) for that input angle are totally bogus

> are you selecting any variants of pico_double implementation?
no

> what compiler & version are you using
GCC 10.3.1, Windows buils

>also did you recreate your build directory after upgrade
Yes of course. 

Did you check if you see similar result?

kilograham, 2025-04-09:
> Did you check if you see similar result?

Not yet; first step is always to see if i can solve the problem without writing any code!

richard9999999999, 2025-04-09:
OK, I understand. In this case reproduction seems to be quite easy, it is enough just to copy that 3 lines of code into begin of main function of any RP2350 project and watch in the debugger the content of those variables.. I am able to reproduce the problem that way easily.

richard9999999999, 2025-04-10:
[pico_float.zip](https://github.com/user-attachments/files/19679096/pico_float.zip)
Above is the pico_float from the recent SDK where the problem was not present. If I replace the pico_float folder in current DEVELOP version by files from that zip, the problem is gone..

richard9999999999, 2025-04-14:
BTW the same pico_float implementation is in current MASTER branch so I suppose the problem is also present in current 2.1.1 Pico SDK release!

kilograham, 2025-04-16:
I wasn't able to reproduce on Linux.

Can you attach an ELF that reproduces this

richard9999999999, 2025-04-17:
OK, the problem is not reproducible as easy as I supposed. 
- I've fount that it seems that angle for which I am getting incorrect sin() and cos() varies across device power ups(!) (but probably not varies across restarts from the debugger)
- I've spend some hours trying to find what triggers the problem. This is the code to reproduce the problem:
```
#include <stdio.h>
#include <math.h>

int main()
{
	bool err = false;
	float roll = 0.18061468f;
	for(int i = -1; i < 25; i++)
	{
		float angle = -roll + ((i + 6) * M_PI / 36);
		float sin_a = sin(angle);
		float cos_a = cos(angle);

		if(sin_a < 0 && cos_a < 0)
		{
			//if you get here, you see incorrect values in sin_a, cos_a variables
			err = true;
			break;
		}
	}	

	//if you comment out following two lines of code, the problem doesn't happen
	float dt_float = 0.5f;	
	int64_t i64 = (int64_t)(dt_float);
}
```
..it doesn't needs anything fancy in MakeList.txt, 
- I use no_flash configuration (probably reproducible also in flash)
- I use cpp compiler GCC10.3.1 on Windows (haven't tried C compiler)
- I am using DEBUG configuration. The problem is reproducible also in RELEASE build however not sure if with above code (optimization can skip those two last lines which triggers the problem.

richard9999999999, 2025-04-17:
[sintst.zip](https://github.com/user-attachments/files/19798265/sintst.zip)

BTW I am not mixing anything: following implementation is called:
```
#ifndef __CORRECT_ISO_CPP_MATH_H_PROTO
  inline _GLIBCXX_CONSTEXPR float
  sin(float __x)
  { return __builtin_sinf(__x); }
```

UKTailwind, 2025-04-18:
Seeing exactly the same issue. I believe this relates to changes in sdk V2.1.0 vs 2.0.0

oyama, 2025-04-18:
I checked out the `2.0.0` tag, `master` branch, and `develop` branch of pico-sdk and reproduced the phenomenon with the following code in each.
It starts returning negative values when x exceeds 1.58f.

main.c
```c
#include <stdio.h>
#include <math.h>
#include "pico/stdlib.h"

int main(void) {
    stdio_init_all();

    float x = 1.6f;
    float y1 = sinf(x);
    float y2 = cosf(x);

    if(y1 >= 0 && y2 >= 0)
        printf("x=%.8f; sin(x)=%.8f; cos(x)=%.8f;  ok\n", x, y1, y2);
    else
        printf("x=%.8f; sin(x)=%.8f; cos(x)=%.8f;  fail\n", x, y2, y2);
     return 0;
}
```

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.13...3.27)

include(pico_sdk_import.cmake)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

project(demo C CXX ASM)
pico_sdk_init()

add_executable(demo main.c)
target_link_libraries(demo pico_stdlib)
pico_add_extra_outputs(demo)
```

```bash
% PICO_SDK_PATH=/Users/oyama/src/github.com/raspberrypi/pico-sdk cmake .. -DPICO_BOARD=pico2 
Using PICO_SDK_PATH from environment ('/Users/oyama/src/github.com/raspberrypi/pico-sdk')
PICO_SDK_PATH is /Users/oyama/src/github.com/raspberrypi/pico-sdk
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from /Users/oyama/src/github.com/raspberrypi/pico-sdk/src/boards/include/boards/pico2.h
Auto-converting non-specific PICO_PLATFORM='rp2350' to 'rp2350-arm-s'
Defaulting platform (PICO_PLATFORM) to 'rp2350-arm-s' based on PICO_BOARD setting.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m33_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m33_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/homebrew/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/homebrew/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/homebrew/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: /opt/homebrew/Frameworks/Python.framework/Versions/3.12/bin/python3.12 (found version "3.12.9") found components: Interpreter
TinyUSB available at /Users/oyama/src/github.com/raspberrypi/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /Users/oyama/src/github.com/raspberrypi/pico-sdk/lib/btstack
[... 58 lines cut ...]
[ 83%] Building ASM object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_float/float_conv_m33.S.obj
[ 84%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_malloc/malloc.c.obj
[ 86%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_atomic/atomic.c.obj
[ 88%] Building CXX object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_cxx_options/new_delete.cpp.obj
[ 89%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_standard_binary_info/standard_binary_info.c.obj
[ 91%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_printf/printf.c.obj
[ 93%] Building ASM object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_crt0/crt0.S.obj
[ 94%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_clib_interface/newlib_interface.c.obj
[ 96%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_stdio/stdio.c.obj
[ 98%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_stdio_uart/stdio_uart.c.obj
[100%] Linking CXX executable demo.elf
[100%] Built target demo
                                                                                                                                                                                      
% picotool load -x demo.uf2                                                                  
Family ID 'rp2350-arm-s' can be downloaded in absolute space:
  00000000->02000000
Loading into Flash:   [==============================]  100%

The device was rebooted to start the application.
 ```

Output:
```
x=1.60000002; sin(x)=-0.02919955; cos(x)=-0.02919955;  fail
```

oyama, 2025-04-18:
I have confirmed that the following code and build with `-O0` will change the execution result between pico-sdk `2.1.0` and `2.1.1`.

main.c
```c
#include <stdio.h>
#include <math.h>
#include "pico/stdlib.h"

int main(void) {
    stdio_init_all();

    float float_value = 0.0f;
    int64_t dummy = (int64_t)float_value;

    float x = 1.565f;
    float y1 = sinf(x);
    float y2 = cosf(x);

    printf("x=%.3f, sin=%.3f, cos=%.3f\n", x, y1, y2);

    return 0;
}
```

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.13...3.27)

include(pico_sdk_import.cmake)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

project(demo C CXX ASM)
pico_sdk_init()

add_executable(demo main.c)
target_compile_options(demo PRIVATE -O0)
target_link_libraries(demo pico_stdlib)
pico_add_extra_outputs(demo)
```

Build
```bash
% PICO_SDK_PATH=/Users/oyama/src/github.com/raspberrypi/pico-sdk cmake .. -DPICO_BOARD=pico2 
Using PICO_SDK_PATH from environment ('/Users/oyama/src/github.com/raspberrypi/pico-sdk')
PICO_SDK_PATH is /Users/oyama/src/github.com/raspberrypi/pico-sdk
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from /Users/oyama/src/github.com/raspberrypi/pico-sdk/src/boards/include/boards/pico2.h
Auto-converting non-specific PICO_PLATFORM='rp2350' to 'rp2350-arm-s'
Defaulting platform (PICO_PLATFORM) to 'rp2350-arm-s' based on PICO_BOARD setting.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m33_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m33_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/homebrew/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/homebrew/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/homebrew/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: /opt/homebrew/Frameworks/Python.framework/Versions/3.12/bin/python3.12 (found version "3.12.9") found components: Interpreter
[... 68 lines cut ...]
[ 91%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_printf/printf.c.o
[ 93%] Building ASM object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_crt0/crt0.S.o
[ 95%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_clib_interface/newlib_interface.c.o
[ 96%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_stdio/stdio.c.o
[ 98%] Building C object CMakeFiles/demo.dir/Users/oyama/src/github.com/raspberrypi/pico-sdk/src/rp2_common/pico_stdio_uart/stdio_uart.c.o
[100%] Linking CXX executable demo.elf
[100%] Built target demo
                                                                                                                                                                                      
% picotool load -x demo.uf2                                                                  
Family ID 'rp2350-arm-s' can be downloaded in absolute space:
  00000000->02000000
Loading into Flash:   [==============================]  100%

The device was rebooted to start the application.
```

Output(2.1.0)
```
x=1.565, sin=1.000, cos=0.006
```

Output(2.1.1)
```
x=1.565, sin=-0.547, cos=-0.837
```

oyama, 2025-04-18:
@richard9999999999 I didn’t understand it at first either—actually, I still don’t.

If I specify `pico_set_float_implementation(TARGET compiler)` or `pico_set_float_implementation(TARGET pico_dcp)` in CMakeLists.txt, it appears normal.
When `pico_set_float_implementation(TARGET pico_vfp)` (RP2350 default?) is used, the problem seems to reproduce.

kilograham, 2025-04-18:
yes, this is nasty... arguably the assembler should report an error... but the problem is dependent on the presence/absence of other bits of code (hence the weird repro)

richard9999999999, 2025-04-18:
Good work!, thank you!

kilograham, 2025-04-21:
merged into develop

PR #2429 (2025-04-21): Fix alignment of lookup table in vfp float implmenetation
fixes #2406
closed 2025-04-21 by kilograham