raspberrypi/pico-sdk #2408: No mention of `PICO_DEOPTIMIZED_DEBUG` in SDK documentation

recursivenomad, 2025-04-10:
The CMake variable `PICO_DEOPTIMIZED_DEBUG` is not mentioned in the Pico SDK documentation, nor the Getting Started guide.

It is an important part of configuration, especially since leaving it unset [results in](https://github.com/raspberrypi/pico-sdk/pull/1621) some debug features not working (at least as of SDK version <1.5.0 when I last checked).  I would have expected it to be listed in [Section 6.1 - Full List of SDK Configuration Variables](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf#cmake-config-variables).

While I'm at it, I went ahead and assembled a list linking to that and other possible contenders from the SDK source that you may or may not want to make note of in the documentation:

*(SDK searched with regex `if.*NOT.*PICO_` )*

- [`PICO_DEOPTIMIZED_DEBUG`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/preload/toolchains/util/set_flags.cmake#L8)
- [`PICO_DEFAULT_BOARD_rp2040`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_platform.cmake#L17)
- [`PICO_DEFAULT_BOARD_rp2350`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_platform.cmake#L20)
- [`PICO_DEFAULT_BOARD_rp2350-arm-s`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_platform.cmake#L23)
- [`PICO_DEFAULT_BOARD_rp2350-riscv`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_platform.cmake#L26)
- [`PICO_DEFAULT_BOARD_host`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_platform.cmake#L29)
- [`PICO_DEFAULT_PLATFORM`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_platform.cmake#L33)
- [`PICO_TOOLCHAIN_DIR`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_toolchain.cmake#L51)
- [`PICO_NO_CMSE`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/preload/toolchains/pico_arm_cortex_m23_gcc.cmake#L11)
- [`PICO_CLIB`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_clib_interface/CMakeLists.txt#L37)
- [`PICO_CMAKE_PRELOAD_PLATFORM_DIR`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/pico_pre_load_platform.cmake#L129)
- [`PICO_PLATFORM_CMAKE_DIR`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/CMakeLists.txt#L8) ?
- [`PICO_PLATFORM_CMAKE_FILE`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/CMakeLists.txt#L11) ?
- [`PICO_TIME_NO_ALARM_SUPPORT`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/host/hardware_timer/CMakeLists.txt#L7)
- [`PICO_DEFAULT_DOUBLE_IMPL`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_double/CMakeLists.txt#L18)
- [`PICO_DEFAULT_BINARY_TYPE`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_standard_link/CMakeLists.txt#L33)
- [`PICO_LINKER_SCRIPT_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_standard_link/CMakeLists.txt#L46)
- [`PICO_FLASH_SIZE_BYTES`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_standard_link/CMakeLists.txt#L50)
- [`PICO_SDK_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/external/pico_sdk_import.cmake#L28)
- [`PICO_SDK_FETCH_FROM_GIT`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/external/pico_sdk_import.cmake#L33)
- [`PICO_SDK_FETCH_FROM_GIT_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/external/pico_sdk_import.cmake#L38)
- [`PICO_SDK_FETCH_FROM_GIT_TAG`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/external/pico_sdk_import.cmake#L43)
- [`PICO_EXAMPLES_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/docs/CMakeLists.txt#L7)
- [`PICO_BTSTACK_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_btstack/CMakeLists.txt#L1)
- [`PICO_CYW43_DRIVER_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_cyw43_driver/CMakeLists.txt#L1)
- [`PICO_LWIP_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_lwip/CMakeLists.txt#L1)
- [`PICO_LWIP_CONTRIB_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_lwip/CMakeLists.txt#L279)
- [`PICO_MBEDTLS_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/pico_mbedtls/CMakeLists.txt#L1)
- [`PICO_TINYUSB_PATH`](https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/tinyusb/CMakeLists.txt#L1)

lurch, 2025-04-26:
> * PICO_BTSTACK_PATH
> * PICO_CYW43_DRIVER_PATH
> * PICO_LWIP_PATH
> * PICO_MBEDTLS_PATH
> * PICO_TINYUSB_PATH

These ones were added in #2405 🙂

lurch, 2025-04-26:
> * PICO_FLASH_SIZE_BYTES

This one has a `PICO_CONFIG:` entry [here](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_flash/include/hardware/flash.h#L53) i.e. it appears in section 5.1 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

> (SDK searched with regex `if.*NOT.*PICO_` )

I decided to use a slightly fancier approach [find_unlabelled_pico_cmake_vars.py.txt](https://github.com/user-attachments/files/19925260/find_unlabelled_pico_cmake_vars.py.txt)  😃 
which found some variables not included in your list above... but I have **NO** idea which of these should be documented, and which are "false positives" i.e. internal-use-only.
* PICO_32BIT
* PICO_BOARD_CMAKE_FILE
* PICO_BOARD_HEADER_FILE
* PICO_BOOT_STAGE2_COMPILE_TIME_CHOICE_NAME
* PICO_BUILD_DOCS
* PICO_CMSIS_CORE_PATH
* PICO_CMSIS_DEVICE
* PICO_COMBINED_DOCS
* PICO_COMPILER_SYSROOT
* PICO_CYW43_SUPPORTED
* PICO_C_COMPILER_IS_CLANG
* PICO_C_COMPILER_IS_GNU
* PICO_DEBUG_INFO_IN_RELEASE
* PICO_DEFAULT_COMPILER
* PICO_DEFAULT_FLASH_SIZE_BYTES
* PICO_MBEDTLS_CONFIG_FILE
* PICO_PLATFORM_PREFIX
* PICO_PREVIOUS_CMAKE_TOOLCHAIN_FILE
* PICO_PREVIOUS_PLATFORM
* PICO_PROMOTE_COMMON_SCOPE_VARS
* PICO_RISCV
* PICO_RP2040
* PICO_RP2350
* PICO_SAVED_PLATFORM
* PICO_SAVED_PLATFORM_PREFIX
* PICO_SDK_INIT_CMAKE_FILE
* PICO_SDK_TESTS_ENABLED
* PICO_SDK_TOP_LEVEL_PROJECT
* PICO_SYMLINK_ELF_AS_FILENAME

recursivenomad, 2025-04-26:
> This one has a PICO_CONFIG: entry [here](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_flash/include/hardware/flash.h#L53) i.e. it appears in section 5.1 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

Actually that one lives in [Section 6.1](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf#cmake-config-variables) 🙂  Good catch, I missed that when drafting my list above.

> I decided to use a slightly fancier approach [find_unlabelled_pico_cmake_vars.py.txt](https://github.com/user-attachments/files/19925260/find_unlabelled_pico_cmake_vars.py.txt) 😃

😁 My reasoning for searching for `if.*NOT.*PICO_` was that would specifically show me CMake options which (probably) have a default value (but imply that the user can override them).  I also do not know which of my list (or yours) are relevant, but a few seem important on first glance.

And just to clarify for the curious, both sections in the documentation exist:
- [Section 5.1 - Full List of SDK Configuration Defines](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf#_full_list_of_sdk_configuration_defines) (SDK `#define` macros)  
  and
- [Section 6.1 - Full List of SDK Configuration Variables](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf#cmake-config-variables) (CMake options, what both our lists are referencing)