raspberrypi/pico-sdk #2417: Using QSPI pins as a GPIO (for example for Blink)

Fantonrko, 2025-04-12:
Hi, I'm using RP2350 without Flash. I managed to run UART on pins SCLK, CS, SD0, SD1, SD2, SD3. But I don't understand how to use them as SIO, for example to blink a LED or Read State

lurch, 2025-04-14:
Sounds like a duplicate of #1799 ?

But pinging @andygpz11 as he has also been using Flash-less RP2350s (on an internal project), so he might have some helpful tips?

andygpz11, 2025-04-14:
@lurch Thanks for the ping.

@Fantonrko I have attached a zip of an _example_ module I have generated by extracting the relevant parts from some source code we are using internally.

To get it building I just added the module to one of the `pico-examples/gpio/...` examples `CMakeLists.txt` file and added a call to `qspi_and_gpio_test_setup_pins_builds()` in that example's `main()` function to make sure the extra module got built.

It does build but it is not currently easy for me to test this.  That said, I would expect this to work for you.

Two things to note although I suspect you are well aware of the first point:

1. The pin function select choices for the QSPI pins are only a small subset of those available for the main bank of GPIO pins.
2. The USB Phy pins can also be used a GPIOs but require a little extra magic even above the QSPI pins.  My example code does not include this since you didn't ask and indeed it further complicates the source code.

Please see attached, good luck and please keep us posted.

[qspi_gpios.zip](https://github.com/user-attachments/files/19738731/qspi_gpios.zip)

d-c-d, 2025-04-14:
thanks @andygpz11 
we also had trouble finding `GPIO_FUNC_SIO` in `rp2040/hardware_structs/include/hardware/structs/io_bank0.h`
it would be nice if pico-sdk provided an interface to `qspi_and_gpio_init` and `qspi_only_set_function`

we have some rp2040's that share the same flash and after disabling the flash - re-use the pins for communication 
( but now that the rp2350 is out this is not required )

Fantonrko, 2025-04-14:
Thanks all! Will check all this info

kilograham, 2026-11-11:
there is an example now, and closing as duplicate of #1799 for sdk functionality

closed 2026-11-11 by kilograham