raspberrypi/pico-sdk #2419: Broken links in readme?

diminDDL, 2025-04-12:
The readme contains a documentation sections with several links. All of them seem to lead to the same https://www.raspberrypi.com/products/raspberry-pi-pico/ page. I don't believe that is intentional.

Should they be replaced with links to the pages/documents they are referring to?

Thank you.

d-c-d, 2025-04-12:
@diminDDL - this is my guess - as it looks like `https://rptl.io` went away
Some of the links likely go to DataSheet PDF files, while other might be going to online pages

perhaps
[Connecting to the Internet with Raspberry Pi Pico W](https://rptl.io/picow-connect)
should be 
[Connecting to the Internet with Raspberry Pi Pico W](https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf)

[Raspberry Pi Pico-Series C/C++ SDK](https://rptl.io/pico-c-sdk) 
should be:
[Raspberry Pi Pico-Series C/C++ SDK](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf)

[Online Raspberry Pi Pico SDK API docs](https://rptl.io/pico-doxygen)
should be:
[Online Raspberry Pi Pico SDK API docs](https://www.raspberrypi.com/documentation/pico-sdk/)

[Raspberry Pi Pico-Series C/C++ SDK](https://rptl.io/pico-c-sdk)
should be:
[Raspberry Pi Pico-Series C/C++ SDK](https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html)

[Raspberry Pi Pico-Series C/C++ SDK](https://rptl.io/pico-c-sdk)
I'm not sure if this should be 
[Raspberry Pi Pico-Series C/C++ SDK](https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html)
   or
[Raspberry Pi Pico-Series C/C++ SDK](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf)

the last link
[Raspberry Pi Pico-series C/C++ SDK](https://rptl.io/pico-c-sdk)
probably
[Raspberry Pi Pico-Series C/C++ SDK](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf)

JamesH65, 2025-04-14:
@brian-rpi Has the link shortener had an issue?

lurch, 2025-04-14:
There was a brief hiccup with our URL-redirector, which has now been fixed. Thanks for the report!

closed 2025-04-14 by lurch