raspberrypi/pico-sdk #2420: localtime_r is missing definition when using the clang compiler

NolanBrad, 2025-04-13:
When compiling using bazel build, and using the clang compiler, localtime_r is missing definition

i.e. in bazelrc
```
build --@pico-sdk//bazel/config:PICO_TOOLCHAIN=clang
build --@pico-sdk//bazel/config:PICO_CLIB=auto
```

Results in error:
_ld.lld: error: undefined symbol: localtime_r
 referenced by datetime.c:11 (/proc/self/cwd/external/pico-sdk+/src/common/pico_util/datetime.c:11)
 datetime.o:(pico_localtime_r) in archive bazel-out/k8-dbg/bin/external/pico-sdk+/src/common/pico_util/libpico_util.a
clang++: error: ld.lld command failed with exit code 1 (use -v to see invocation)_

localtime_r has is declared in pico-sdk/src/rp2_common/pico_clib_interface/include/llvm_libc/time.h
`struct tm* localtime_r(const time_t* timer, struct tm* buf);`

I am using the pico_aon_timer library, specifically  the aon_timer_set_time function

It looks as though the clib for clang doesnt support localtime_r.  See https://libc.llvm.org/headers/time.html

lurch, 2025-04-14:
ping @armandomontanez

armandomontanez, 2025-04-14:
@petrhosek FYI

petrhosek, 2025-04-15:
`localtime_r` is under development, see https://github.com/llvm/llvm-project/pull/110363.