raspberrypi/pico-sdk #2428: Can the phase of the PIO clock of the RP2040 be adjusted?

a0k1i2n3g, 2025-04-17:
Can the PIO of the RP2040 set a fixed additional phase offset for the clock of each state machine? In this way, multiple PIOs can be used to sample the same GPIO to achieve the so - called oversampling. With this, the RP2040 can replace many CPLDs used for sampling, and the data processing is done through C - language programs. Just thinking about it gets me excited!

peterharperuk, 2025-04-17:
@Wren6991 might know?

Wren6991, 2025-07-18:
The only way of adjusting the phase is to reset it, which you can do for multiple state machines simultaneously to synchronise them.

If you want multiple samples of a pin with a short sampling interval then the simplest approach is normally just to run one state machine at full speed and capture multiple back-to-back samples into the input shift register.

Note there is no finer granularity available than one system clock cycle. The "fractional" clock division is actually just a fixed-point accumulator which is generating an enable for the system clock.