raspberrypi/pico-sdk #2448: Build errors for pioasmBuild with GCC 15.1.1

dpgeorge, 2025-05-02:
GCC 15 is out (eg on Arch Linux), and building MicroPython Pico W and Pico 2 W (and probably anything that uses cyw43) now gives an error, like this:
```
<checkout latest micropython repo>
$ cd ports/rp2
$ make BOARD=RPI_PICO_W

...

-- The CXX compiler identification is GNU 15.1.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: micropython/ports/rp2/build-RPI_PICO_W/pioasm                     
[  2%] Performing build step for 'pioasmBuild'
In file included from micropython/lib/pico-sdk/tools/pioasm/gen/parser.hpp:52,                     
                 from micropython/lib/pico-sdk/tools/pioasm/pio_assembler.h:11,                     
                 from micropython/lib/pico-sdk/tools/pioasm/main.cpp:8:                     
micropython/lib/pico-sdk/tools/pioasm/pio_types.h:290:5: error: ‘uint8_t’ does not name a type                     
  290 |     uint8_t used_gpio_ranges = 0;
      |     ^~~~~~~
micropython/lib/pico-sdk/tools/pioasm/pio_types.h:19:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
   18 | #include "pio_enums.h"
  +++ |+#include <cstdint>
   19 | 
In file included from micropython/lib/pico-sdk/tools/pioasm/pio_assembler.h:12:                     
micropython/lib/pico-sdk/tools/pioasm/output_format.h:83:9: error: ‘uint8_t’ does not name a type                     
   83 |         uint8_t used_gpio_ranges;
      |         ^~~~~~~
micropython/lib/pico-sdk/tools/pioasm/output_format.h:1:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
  +++ |+#include <cstdint>
    1 | /*
make[6]: *** [CMakeFiles/pioasm.dir/build.make:75: CMakeFiles/pioasm.dir/main.cpp.o] Error 1
make[5]: *** [CMakeFiles/Makefile2:87: CMakeFiles/pioasm.dir/all] Error 2
make[4]: *** [Makefile:136: all] Error 2
make[3]: *** [pico-sdk/src/rp2_common/pico_cyw43_driver/CMakeFiles/pioasmBuild.dir/build.make:95: pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/src/pioasmBuild-stamp/pioasmBuild-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:2459: pico-sdk/src/rp2_common/pico_cyw43_driver/CMakeFiles/pioasmBuild.dir/all] Error 2
make[1]: *** [Makefile:91: all] Error 2
make[1]: Leaving directory 'micropython/ports/rp2/build-RPI_PICO_W'                     
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:65: all] Error 1
```
I'm not sure exactly what changed in GCC 15, but applying the above suggestions to `#include <cstdint>` gets the build working again.

lurch, 2025-05-06:
@liamfraser I guess we also ought to update https://github.com/raspberrypi/pico-sdk/blob/master/.github/workflows/multi-gcc.yml to build with GCC 15 ? Or is that dependant on https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads getting the new GCC version?

projectgus, 2025-05-09:
If anyone else is curious what happened here, it looks like this: 
https://gcc.gnu.org/gcc-15/porting_to.html#header-dep-changes

kilograham, 2025-05-20:
merged into develop

PR #2468 (2025-05-20): Fix pioasm build failure caused by GCC 15 changes (#2448)
Fixes #2448 .
Inserted missing `<cstdint>` in several files to make it possible to build `pioasm` on GCC 15 again. Tested on latest Arch Linux with GCC 15.1.1.
closed 2025-05-20 by kilograham