raspberrypi/pico-sdk #2449: Typo for fractional part of `lposc_freq_khz`?

tommy-gilligan, 2025-05-02:
Currently:

https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/src/rp2_common/hardware_powman/powman.c#L72

I think this line should be referencing `lposc_freq_hz` instead of `lposc_freq_khz`

Given an input value of `lposc_freq_hz = 32768`
`lposc_freq_khz` will be `32768 / 1000 = 32`
This is the integer part
Fractional part cannot be derived from this

I am not sure if the rest of the maths makes sense either.

kilograham, 2025-05-02:
Please feel free to make a PR (for the khz->hz), i think the `65536 / 1000` is good as is, because otherwise you'd be multiplying by 655 not 655.36

tommy-gilligan, 2025-05-02:
Misread the size of variable.  Without parentheses is better.

lurch, 2025-05-06:
Fix merged into `develop`

closed 2025-05-06 by lurch