raspberrypi/pico-sdk #2455: What is the maximum frequency of the full-amplitude square wave output by the GPIO of RP2040?

a0k1i2n3g, 2025-05-05:
When I overclock the RP2040 to 400 MHz and use PWM or SIO to toggle a pin to output a 200 MHz square wave, the peak-to-peak value is only a pitiful 200 mV and fails to reach the full amplitude of 3 V. Even without overclocking, when using a CPU frequency of 200 MHz to output a 100 MHz square wave, the peak-to-peak value is only about 500 mV. The full amplitude of 3 V can only be achieved when the frequency is reduced to 50 MHz. Is this caused by software issues on my side or insufficient hardware driving capability of the RP2040?

Can it be improved through software, or can a transistor be added to amplify the amplitude?

kilograham, 2025-05-05:
This is a better question for the forums: https://forums.raspberrypi.com/viewforum.php?f=143

lurch, 2025-05-06:
And I don't know if it answers any of your questions or not, but see Chapter 5 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf