raspberrypi/pico-sdk #2459: BTstack implementation by Blue Kitchen Lacks Vital api/data

marcosgabrielserrano, 2025-05-08:
Important events such as: GATT_EVENT_CHARACTERISTIC_QUERY_RESULT or GATT_EVENT_CHARACTERISTIC_VALUE_QUERY_RESULT do not return the connection_handle, and instead return the attribute or value handles.

Because of this, _you cannot correlate reading value of a characteristic to a specific connection_.

The problem that arises is that during multiple connections with same service/characteristic/handle_value, you can't track from which connection you are reading. Therefore you might think you are reading from client A, but really be reading the same char from client B or C.

As such the only way I see around this right now is to only allow a single read at a time and track who is performing the read with overhead, then allowing the next client to send a read characteristic request. This is quite the bottleneck.

Curiously I noticed that `gatt_client_read_value_of_characteristic_using_value_handle_with_context` exists, but I haven't found an equivalent `gatt_event_service_query_result_get_handle` function in the SDK rendition of btstack.

In terms of solutions, is there a version of btstack that passes the connection handle to the above events, or provides a `gatt_client_get_context_for_handle` to work with the `_with_context` function?

Additionally, my basic knowledge of the connection_id, is that it is session specific, and may not map to the hci connection handle. (basically it changes? This could be another of my misunderstandings?)

Am I missing something?

peterharperuk, 2025-05-08:
@mringwal Any idea about this?

mringwal, 2025-05-08:
Hi there. This is a general question about BTstack and could preferably asked on the BTstack mailing list.

The first field in GATT_EVENT_CHARACTERISTIC_VALUE_QUERY_RESULT and GATT_EVENT_CHARACTERISTIC_VALUE_QUERY_RESULT is the connection handle. It also contains the value_handle for the Characteristic Value itself.

However, there's a different problem implementing generic services clients, as we need to dispatch the events to the corresponding service client implementations. For this, we've added the service_id and connection_id fields. They are not used by the GATT Client and passed back in the corresponding events. You are free to use them in your application code. We use them in the gatt_service_client.h implementation to support multiple instances of a single service type (e.g. more than one battery service).

peterharperuk, 2025-05-08:
Thanks @mringwal !

closed 2025-05-08 by peterharperuk

marcosgabrielserrano, 2025-05-08:
@mringwal Could you provide some more clarification on how the GATT_EVENT_CHARACTERISTIC_VALUE_QUERY_RESULT field handle is the connection handle? My understanding is the field handle is specific to the characteristic value handle. 

How would I extract the connection handle and how would I extract the value handle if this field handle has both?

mringwal, 2025-05-08:
The first field/parameter called "handle" is the connection handle.

On the current BTstack master, we have

/**
 * @format H222LV
 * @param handle
 * @param service_id
 * @param connection_id
 * @param value_handle
 * @param value_length
 * @param value
 */
#define GATT_EVENT_CHARACTERISTIC_VALUE_QUERY_RESULT             0xA5u

The service_id and connection_id have been added in Aug 2024, the other fields have been there since 2015.

You can get the connection handle with gatt_event_characteristic_value_query_result_get_handle(packet).
You can get the value handle with gatt_event_characteristic_value_query_result_get_value_handle(packet).

marcosgabrielserrano, 2025-05-08:
@mringwal 

First I would like to say thank you so much for your information. 

Second: just to confirm so I can identify which connection's char value query via both connection_id AND the handle correct?

Third fyi, I was misled by AI on this and had to wrangle it to admit it was wrong. It looks like AI didn't like the naming convention:

[image]

Users of AI beware. This took me through a loop.

mringwal, 2025-05-08:
You can use connection_id and service_id - bot h16-bit values - however you seem fit if you use gatt_client.h API directly.

You should be able to be able to match results for multiple parallel queries to different devices and/or characteristics by reading the connection handle (in handle field) as well as value handle (in value_handle).

AI can be helpful, but I found it most valuable as a better google search which gives me hints on where to look / read the details. For BTstack events, the btstack_defines.h is important and it's unclear, you can use "find usage" in an IDE to see how the event is constructed.

marcosgabrielserrano, 2025-05-08:
@mringwal 

By the handle/connection_id usage I primarily mean that in the packet handler itself, how would the handler identify which connection's char value it's reading in a case where there are multiple concurrent connections using same service/chars etc....

mringwal, 2025-05-08:
Well. It’s up to you how to do that. By providing the optional values, your code has sufficient information to match its own requests. (For multiple requests to the same connection handle, both your code and remote need to support Extended ATT)Sent from my iPhone