raspberrypi/pico-sdk #2460: PIO ISR to CPU never works (PIO to DMA or any other way)

angelsix, 2025-05-08:
RP2350B (Waveshare PICO2-XXL) for testing. Attached code, shown below also. 

Just trying to get the most basic PIO to array code working. 

Tried many ways:

- adding delays
- forcing clearing buffer
- different dreq settings
- difference Pi board
- IN command with pins, null or other inputs (x, y)
- PIO 0 and 1
- With/without interrupts
- everything I can find on any documentation

All that ever happens is as soon as the FIFO is full (3 16bits) it hangs.

For clarity, the program runs identically if you have no DMA setup at all. So, it's clearly the DMA not doing anything at all. The DMA full handler is never fired either.

For simplicity I just toggle a pin using side-set to see when the PIO stops running due to push being a full FIFO

[image]

I cannot find a single example anywhere online or the docs of how to actually use the IN command to a DMA or the IN command at all really. 

Anyone see what's wrong?

test.cpp

```
/*
      IMPORTANT: Make sure to edit C:\Users\conta\.pico-sdk\sdk\2.1.1\src\boards\include\boards\pico2.h
                changing #define PICO_RP2350A 1 to #define PICO_RP2350A 0
                as CMakeLists.txt set(PICO_BOARD pico2 CACHE STRING "Board type") makes it use pico2.h
                and if not changed to B variant, the upper 32-47 pins are not accessible

*/

#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/pio.h"
#include "hardware/dma.h"

// Our assembled programs:
// Each gets the name <pio_filename.pio.h>
#include "test.pio.h"

// Pins
#define PIN_LED 25
#define PIN_DEBUG 32

#define BUFFER_SIZE 160

unsigned short buffer[BUFFER_SIZE];

uint sm = 2;
uint offset;

int dma_chan;
dma_channel_config dma_config;

PIO pio = pio1;

// Called when buffer full
void dma_full_handler()
{
    // Clear the interrupt request.
    dma_hw->ints0 = 1u << dma_chan;

    dma_channel_configure(
        dma_chan,               
        &dma_config,          
        &buffer,            
        &pio->rxf[sm], 
        BUFFER_SIZE,     
        true                        // Start immediately.
    );
}

int main()
{
    // Basic setup
    stdio_init_all();

     // Set base to higher pins 16-47
     pio_set_gpio_base(pio, 16);

     offset = pio_add_program(pio, &test_program);
 
     pio_sm_config config = test_program_get_default_config(offset);
     sm_config_set_sideset_pins(&config, PIN_DEBUG);
     pio_sm_set_consecutive_pindirs(pio, sm, PIN_DEBUG, 1, true);
     pio_sm_init(pio, sm, offset, &config);
     pio_gpio_init(pio, PIN_DEBUG);
     sm_config_set_in_shift(&config, true, true, 32);
 
     pio_sm_init(pio, sm, offset, &config);
 
     pio_sm_set_enabled(pio, sm, true);

    // Get unused DMA channel
    dma_chan = dma_claim_unused_channel(true);

    // Congigure
    dma_config = dma_channel_get_default_config(dma_chan);                         // default configs
    channel_config_set_transfer_data_size(&dma_config, DMA_SIZE_16);                  // 16-bit txfers 
    channel_config_set_read_increment(&dma_config, false);                            // no read incrementing
    channel_config_set_write_increment(&dma_config, true);                            // yes write incrementing
    channel_config_set_dreq(&dma_config, pio_get_dreq(pio, sm, true)) ;  // Pace the FIFO to the writing of data

    dma_channel_configure(
        dma_chan,                // Channel to be configured
        &dma_config,          
        &buffer,           
        &pio->rxf[sm], 
        BUFFER_SIZE,     
        true                        // Start immediately.
    );

    // Configure the processor to run dma_handler() when DMA IRQ 0 is asserted
    irq_set_exclusive_handler(DMA_IRQ_0, dma_full_handler);
    irq_set_enabled(DMA_IRQ_0, true);
    
    // Tell the DMA to raise IRQ line 0 when the channel finishes a block
    dma_channel_set_irq0_enabled(dma_chan, true);

    // Turn on LED to let us know we are live
    gpio_put(PIN_LED, true);

    while (true) { 
        sleep_ms(100);
     }
}

```

test.pio

```
.program test
.side_set 1 opt 

.wrap_target                      ; Program wraps to here
    in null, 4 side 1 [7]
    in null, 4 side 1 [7]
    in null, 4 side 0 [7]
    in null, 4 side 0 [7]
.wrap
```

[Reading from ISR.zip](https://github.com/user-attachments/files/20102511/Reading.from.ISR.zip)

Built using Raspberry Pi Pico extension in VS Code.

lurch, 2025-05-08:
> I cannot find a single example anywhere online or the docs of how to actually use the IN command to a DMA or the IN command at all really.

Does the `logic_analyser` example at https://github.com/raspberrypi/pico-examples?tab=readme-ov-file#pio do something similar to what you're looking for?

angelsix, 2025-05-08:
I figured it out, I had `pio_get_dreq` last parameter set to true not false (for RXF not TXF).

However, now I have done that, I presume I am getting some kind of FIFO bottleneck as I send 160 16bit values with 33us between them, but then I see these pauses periodically, until the entire thing stops after around 58 milliseconds and I have to reset it all. It never recovers.

[image]

Any ideas how to improve that or what the cause might be?

closed 2025-05-08 by kilograham

angelsix, 2025-05-08:
You can see here a serial dump of the buffer prior to running the pio / DMA. Then right after it all gets wiped to 0s

[image]

angelsix, 2025-05-08:
A quick update also. Doing this `mov isr, ~null` instead of `in y, 4` results in the actual ~null data passing through successfully, while all the in calls seem to be ignored.

```
.program test
.side_set 1 opt                   

    set y, 28
.wrap_target                      ; Program wraps to here
    mov isr, ~null
    in y, 4 side 1 [7]
    in y, 4 side 1 [7]
    in y, 4 side 0 [7]
    in y, 4 side 0 [7]
.wrap
```

[image]

So it seems it relates directly to the IN command

angelsix, 2025-05-08:
16bit pushes, sent using: `y = 0001 1111 (31)`

```
in y, 4 =  0000 0000 0000 0000 (0)
in y, 5 =  1111 0000 0000 0000 (61440)
in y, 6 =  1100 0000 0000 0000 (49152)
in y, 7 =  1111 0111 1101 1000 (63448)
in y, 8 =  0000 0000 0000 0000 (0)
in y, 9 =  1100 0000 0000 0000 (49152)
in y, 10 = 1111 0000 0000 0000 (61440)
in y, 11 = 0111 1100 0001 1110 (31774)
in y, 12 = 0001 1111 0000 0000 (7936)
in y, 13 = 0000 0111 1100 0000 (1984)
in y, 14 = 0000 0001 1111 0000 (496)
in y, 15 = 0000 0000 0111 1100 (124)
in y, 16 = 0000 0000 0000 0000 (0)
```

So we can see from 16 to 9 it basically is shifting the 11111 (31) over 2 to the left each time.

The data is in there somewhere, but just not how it should be.

If I change the `DMA_SIZE_16` to `DMA_SIZE_32` and leave everything else the same (the buffer being 16bits per entry for example), then do in y, 16. You see the data correctly.

[image]

So seems like a bug with DMA_SIZE_16?

peterharperuk, 2025-05-08:
The dma read and write addresses must be aligned with the transfer size.

angelsix, 2025-05-08:
> __attribute__((aligned(4)))
> 
> I've never used DMA_SIZE_16 but you might have to take care that data is left-justified in the fifo. There is an example using DMA_SIZE_8 in the SDK which could probably be made to work with DMA_SIZE_16 https://github.com/raspberrypi/pico-examples/blob/master/pio/uart_pio_dma/uart_pio_dma.c#L169

Thanks I'll see what I can figure out in the morning on it

angelsix, 2025-05-09:
So, looks more like a bug as far as I can tell (Would really help if the SDK even had an example of using DMA_SIZE_16 considering the devs are confident there are no bugs, but nobody seems to of ever even used, tried or done DMA_SIZE_16 in any official docs or this GitHub I can find anywhere to confirm as such).

 - DMA_SIZE_16
 - Array aligned or not makes no difference
 - Read address `(io_rw_16*)&pio->rxf[sm] + 1` to read the far right 16bit not empty front 32bit part
 - Disable auto-push
 
Now... if we push 16 bits, we get all the data, but every other byte is 0s.

```
    in x, 4
    in x, 4 
    in x, 4
    in x, 3 
    push
```

```
65535 0 61166 0 56797 0 52428 0 48059 0 43690 0 39321 0 34952 0 30583 0 26214 0 21845 0 17476 0 13107 0 8738 0 4369 0 0 0 65535 0 61166 0 56797 0 52428 0 48059 0 43690 0 39321 0 34952 0 30583 0 26214 0 21845 0 17476 0 13107 0 8738 0 4369 0 0 0 65535 0 61166 0 56797 0 52428 0 48059 0 43690 0 39321 0 34952 0 30583 0 26214 0 21845 0 17476 0 13107 0 8738 0 4369 0 0 0 65535 0 61166 0 56797 0 52428 0 48059 0 43690 0 39321 0 34952 0 30583 0 26214 0 21845 0 17476 0 13107 0 8738 0 4369 0 0 0 65535 0 61166 0 56797 0 52428 0 48059 0 43690 0 39321 0 34952 0 30583 0 26214 0 21845 0 17476 0 13107 0 8738 0 4369 0 0 0 
```

Now... push 15 bits instead of 16. The data in now equals the data out with the LSB as zero.

I send in 1111, then 1110, then 1101 etc... counting down 4bit values repeated 4 times. However, as the last 4bit I send only 3, that then results in 1111 1111 1111 1110 for the first one and so on. Which is what gets sent in.

```
    in x, 4
    in x, 4 
    in x, 4
    in x, 3 
    push
```

```
65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 65534 56796 48058 39320 30582 21844 13106 4368 61166 52428 43690 34952 26214 17476 8738 0 
```

Conclusion so far... setup seems right, works exactly as expected, but as soon as you push the actual 16th bit, the DMA seems to jump a full address.

Thoughts?

angelsix, 2025-05-09:
Changing the code a bit to just directly push 15 or 16 bits shows its perhaps not even that simple. It might still be skewing data. 

X is loaded with 160 and counts down.

`in x, 16; push`

```
159 0 158 0 157 0 156 0 155 0 154 0 153 0 152 0 151 0 150 0 149 0 148 0 147 0 146 0 145 0 144 0 143 0 142 0 141 0 140 0 139 0 138 0 137 0 136 0 135 0 134 0 133 0 132 0 131 0 130 0 129 0 128 0 127 0 126 0 125 0 124 0 123 0 122 0 121 0 120 0 119 0 118 0 117 0 116 0 115 0 114 0 113 0 112 0 111 0 110 0 109 0 108 0 107 0 106 0 105 0 104 0 103 0 102 0 101 0 100 0 99 0 98 0 97 0 96 0 95 0 94 0 93 0 92 0 91 0 90 0 89 0 88 0 87 0 86 0 85 0 84 0 83 0 82 0 81 0 80 0 
```

`int x, 15; push` doubles the values 

```
318 316 314 312 310 308 306 304 302 300 298 296 294 292 290 288 286 284 282 280 278 276 274 272 270 268 266 264 262 260 258 256 254 252 250 248 246 244 242 240 238 236 234 232 230 228 226 224 222 220 218 216 214 212 210 208 206 204 202 200 198 196 194 192 190 188 186 184 182 180 178 176 174 172 170 168 166 164 162 160 158 156 154 152 150 148 146 144 142 140 138 136 134 132 130 128 126 124 122 120 118 116 114 112 110 108 106 104 102 100 98 96 94 92 90 88 86 84 82 80 78 76 74 72 70 68 66 64 62 60 58 56 54 52 50 48 46 44 42 40 38 36 34 32 30 28 26 24 22 20 18 16 14 12 10 8 6 4 2 0 
```

So it seems still an alignment issue. I have this issue tagged and posted on the normal forum too.

angelsix, 2025-05-09:
Got it :D Auto-push was still enabled somehow. I just cleaned everything (CMake, build folder etc...), removed manual push and left in auto-push (tried both ways) and pushed code again and this is the solution that works for 16bit transfers:

pio file simply pushes 16 bits in one go or in chunks to suit (tested sending 4s or 16s)

```
    in x, 16
```

DMA configuration

```
pio_sm_config config = test_program_get_default_config(offset);
     sm_config_set_sideset_pins(&config, PIN_DEBUG);
     pio_sm_set_consecutive_pindirs(pio, sm, PIN_DEBUG, 1, true);
     pio_sm_init(pio, sm, offset, &config);
     pio_gpio_init(pio, PIN_DEBUG);
     sm_config_set_in_shift(&config, true, true, 16);
     sm_config_set_in_pins(&config, PIN_D0);
 
     pio_sm_init(pio, sm, offset, &config);
 
     pio_sm_set_enabled(pio, sm, true);

    // Get unused DMA channel
    dma_chan = dma_claim_unused_channel(true);

    // Congigure
    dma_config = dma_channel_get_default_config(dma_chan);                         // default configs
    channel_config_set_transfer_data_size(&dma_config, DMA_SIZE_16);                  // 16-bit txfers 
    channel_config_set_read_increment(&dma_config, false);                            // no read incrementing
    channel_config_set_write_increment(&dma_config, true);                            // yes write incrementing
    channel_config_set_dreq(&dma_config, pio_get_dreq(pio, sm, false)) ;  // Pace the FIFO to the writing of data

    dma_channel_configure(
        dma_chan,                // Channel to be configured
        &dma_config,          
        &buffer,           
        (io_rw_16*)&pio->rxf[sm] + 1, 
        BUFFER_SIZE,     
        true                        // Start immediately.
    );

    // Configure the processor to run dma_handler() when DMA IRQ 0 is asserted
    irq_set_exclusive_handler(DMA_IRQ_0, dma_full_handler);
    irq_set_enabled(DMA_IRQ_0, true);
    
    // Tell the DMA to raise IRQ line 0 when the channel finishes a block
    dma_channel_set_irq0_enabled(dma_chan, true);
```

Loaded X with 160 and looped

```
159 158 157 156 155 154 153 152 151 150 149 148 147 146 145 144 143 142 141 140 139 138 137 136 135 134 133 132 131 130 129 128 127 126 125 124 123 122 121 120 119 118 117 116 115 114 113 112 111 110 109 108 107 106 105 104 103 102 101 100 99 98 97 96 95 94 93 92 91 90 89 88 87 86 85 84 83 82 81 80 79 78 77 76 75 74 73 72 71 70 69 68 67 66 65 64 63 62 61 60 59 58 57 56 55 54 53 52 51 50 49 48 47 46 45 44 43 42 41 40 39 38 37 36 35 34 33 32 31 30 29 28 27 26 25 24 23 22 21 20 19 18 17 16 15 14 13 12 11 10 9 8 7 6 5 4 3 2 1 0 
```

Ultimately the only change from the very first example is this `(io_rw_16*)&pio->rxf[sm] + 1, `. Thanks @peterharperuk for the guidance. 

Perhaps this example or note about address can be used in some documentation or in pico-examples for others to have an easy rout to 8/16bit DMA. Seems it is not obviously mentioned anywhere without a lot of digging and throws up many issues (mainly 0s appearing as no transfers happening).

Glad its resolved.

lurch, 2025-05-09:
> Glad its resolved.

Thank you for confirming that it isn't an SDK bug.