raspberrypi/pico-sdk #2462: clock_gpio_init_int_fracX - input souce clk_usb use sys clock, clk_adc works correctly

rudafm, 2025-05-12:
Verified on scope : clock_gpio_init_int_frac16(15, clk_usb, 2, 0); // ClkOut 24 Mhz - give 75 MHz clock output.
When I use clk_adc it give me 24 MHz - OK.
/ SDK 2.1.1 /.
Regards,
RK

lurch, 2025-05-12:
Some more details would be really appreciated, e.g. what behaviour did you expect and what behaviour are you getting? What hardware are you using? And if your could provide the full code for a simple example that reproduces your problem (rather than just a single line of code), that'd be very helpful.

peterharperuk, 2025-05-30:
You shouldn't use clk_usb with this API. You're actually dividing clk_sys and clk_adc was actually dividing the usb clock. The following will give you the result you're expecting.

```
clock_gpio_init_int_frac16(15, CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLK_USB, 2, 0)
```

It's a bit confusing but the docs say "The source clock. See the register field CLOCKS_CLK_GPOUT0_CTRL_AUXSRC for a full list. The list is the same for each GPOUT clock generator."

So for rp2350 this is what you have to pass in.

```
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLKSRC_PLL_SYS _u(0x0)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLKSRC_GPIN0 _u(0x1)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLKSRC_GPIN1 _u(0x2)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB _u(0x3)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB_PRIMARY_REF_OPCG _u(0x4)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_ROSC_CLKSRC _u(0x5)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_XOSC_CLKSRC _u(0x6)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_LPOSC_CLKSRC _u(0x7)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLK_SYS _u(0x8)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLK_USB _u(0x9)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLK_ADC _u(0xa)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLK_REF _u(0xb)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLK_PERI _u(0xc)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_CLK_HSTX _u(0xd)
#define CLOCKS_CLK_GPOUT0_CTRL_AUXSRC_VALUE_OTP_CLK2FC _u(0xe)

```

closed 2025-05-30 by peterharperuk