raspberrypi/pico-sdk #2469: best_effort_wfe_or_timeout hangs indefinitely

writeonlymemory, 2025-05-16:
I don't exactly know why yet but I am seeing something odd with `best_effort_wfe_or_timeout`.

Given this program:

```
int main() {
  const absolute_time_t deadline = make_timeout_time_ms(1000);
  best_effort_wfe_or_timeout(deadline);
  best_effort_wfe_or_timeout(deadline);
}
```

The second call never returns (the first does, immediately).

If I add an alarm (even far in the future):

```
int main() {
  add_alarm_at(make_timeout_time_ms(5000), do_nothing, NULL, true);
  const absolute_time_t deadline = make_timeout_time_ms(1000);
  best_effort_wfe_or_timeout(deadline);
  best_effort_wfe_or_timeout(deadline);
}
```

The second call to `best_effort_wfe_or_timeout` works as expected.

I'm on a Pico 2W, using SDK 2.1.1.

kilograham, 2025-05-16:
can you try on the `develop` branch

writeonlymemory, 2025-05-16:
Just tried and it seems to have the same behaviour I'm afraid.

kilograham, 2025-05-19:
i tried this with both release and debug builds, but couldn't reproduce on GCC 13.2

@peterharperuk / @will-v-pi  wanna try?

@writeonlymemory what compiler? how r u running (e.g. debugger etc.?

writeonlymemory, 2025-05-19:
Thanks for looking at it.

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:13.2.rel1-2) 13.2.1 20231009
```

Great question about debugging: this jogged my memory about #1977 which I was also hitting. (I am using the debug probe and openocd to flash but not actively debugging the core - just using printf debugging.)

If I power cycle the part after flashing (and getting into this hang) it starts up correctly. So I think I'm just hitting the same oddity at startup as in #1977 

I'll leave it to you to if you want to dupe this into it.