raspberrypi/pico-sdk #2471: powman_timer_use_gpio writing wrong value to ext_time_ref register?

czietz, 2025-05-17:
I wonder how this can be right:

https://github.com/raspberrypi/pico-sdk/blob/550aa09624391dfbc3afae55ede3e008a3879614/src/rp2_common/hardware_powman/powman.c#L100-L111

According to the RP2350 datasheet:

[image]

... ext_time_ref does not directly take the GPIO number (12, 20, 14, 22), but is an enum (0 ... 3). However, the function above writes the GPIO number directly into the register.

lurch, 2025-05-17:
https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2350/hardware_regs/include/hardware/regs/powman.h#L1180 seems to agree - looks like you might have found a bug? It's probably just reading the lower 2 bits of the GPIO value, so trying to set either 12 or 20 will result in GPIO12 being selected, and trying to set 14 or 22 will result in GPIO14 being selected.

kilograham, 2025-07-20:
fix merged into develop

commit e62cd02a (2025-07-16): Powman fix (#2499)

* Fix powman_timer_use_gpio

Fixes #2471

* Fix docs for clock_gpio_init etc for RP2350

You can also use gpios 13 and 15 on RP2350

* macro-ify the mapping from GPI to ext_time_ref_source - note i kept this private for now, as it's a bit murky

---------
closed 2025-07-20 by kilograham