raspberrypi/pico-sdk #2476: pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h has extgraneous characters

tomdean1939, 2025-05-19:
pico-sdk/src/rp2_common/hardware_base/include/hardware/address_mapped.h:147 has 'aaddress_mapped
' after '}'

Removing these characters removes the error.

Tom Dean

lurch, 2025-05-19:
Both `master` https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_base/include/hardware/address_mapped.h#L147 and `develop` https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_base/include/hardware/address_mapped.h#L147 look fine to me, so it looks like you've either accidentally edited that file, or you've got some kind of disk corruption?

tomdean1939, 2025-05-20:
Thank you, fsck shows no problems...
Please close.

closed 2025-05-20 by lurch