raspberrypi/pico-sdk #2479: CRT0.s is tailored for a cm0 not a cm33

tbandtg, 2025-05-21:
Working on fault handlers for my project(2350) noticed that busfault and usage fault are not available.

from the normal cortex m33.s file below is my patch 

--- C:\Work\old.txt
+++ C:\Work\new 3.txt
@@ -29,22 +29,22 @@
 .global __vectors, __VECTOR_TABLE
 __VECTOR_TABLE:
 __vectors:
-.word __StackTop
-.word _reset_handler
-.word isr_nmi
-.word isr_hardfault
-.word isr_invalid // Reserved, should never fire
-.word isr_invalid // Reserved, should never fire
-.word isr_invalid // Reserved, should never fire
-.word isr_invalid // Reserved, should never fire
-.word isr_invalid // Reserved, should never fire
-.word isr_invalid // Reserved, should never fire
-.word isr_invalid // Reserved, should never fire
-.word isr_svcall
-.word isr_invalid // Reserved, should never fire
-.word isr_invalid // Reserved, should never fire
-.word isr_pendsv
-.word isr_systick
+    .word   __StackTop          // Initial Main Stack Pointer
+    .word   _reset_handler      // Reset Handler
+    .word   isr_nmi             // NMI Handler
+    .word   isr_hardfault       // HardFault Handler
+    .word   isr_memmanage       // MemManage Fault Handler
+    .word   isr_busfault        // BusFault Handler
+    .word   isr_usagefault      // UsageFault Handler
+	.word isr_invalid // Reserved, should never fire
+	.word isr_invalid // Reserved, should never fire
+	.word isr_invalid // Reserved, should never fire
+	.word isr_invalid // Reserved, should never fire
+    .word   isr_svcall          // SVCall Handler
+    .word   isr_debugmon        // Debug Monitor Handler
+	.word isr_invalid // Reserved, should never fire
+    .word   isr_pendsv          // PendSV Handler
+    .word   isr_systick         // SysTick Handler
 #if PICO_NO_STORED_VECTOR_TABLE && !PICO_NO_FLASH // note in no flash binary, we only have the single RAM vector table anyway
 #if PICO_NO_RAM_VECTOR_TABLE
 #error Can't specify PICO_NO_STORED_VECTOR_TABLE and PICO_NO_RAM_VECTOR_TABLE
@@ -164,13 +164,15 @@
     bkpt #0
 .endm
 
-// these are separated out for clarity
-decl_isr_bkpt isr_invalid
-decl_isr_bkpt isr_nmi
-decl_isr_bkpt isr_hardfault
-decl_isr_bkpt isr_svcall
-decl_isr_bkpt isr_pendsv
-decl_isr_bkpt isr_systick
+    decl_isr_default isr_nmi
+    decl_isr_default isr_hardfault
+    decl_isr_default isr_memmanage
+    decl_isr_default isr_busfault
+    decl_isr_default isr_usagefault
+    decl_isr_default isr_svcall
+    decl_isr_default isr_debugmon
+    decl_isr_default isr_pendsv
+    decl_isr_default isr_systick
 
 .global __default_isrs_end
 __default_isrs_end:

tbandtg, 2025-05-21:
Should be noted that my solution does not work because even when I trigger what should be a usagefault I get a hard fault.

kilograham, 2025-05-21:
did you configure enable the specific exception you wanted on the M33 - they all map to hardfault by default.

tbandtg, 2025-05-22:
Thats awesome if I could also make the suggestion of having a cmake variable to include your own crt0.s and newlib_interface.c

reason is that I am staining my stack on this project and I would like to do it at the earliest I can.

kilograham, 2025-06-12:
closing as #2482 is merged into develop and that is the original intent of this issue - please open another issue for anything remaining

closed 2025-06-12 by kilograham