raspberrypi/pico-sdk #2481: Unexpected parameter change in macro between different branches

alt-0191, 2025-05-21:
I encountered an unexpected API change when working on my project but different branches of the same SDK version:  

In the old_branch, the `TUD_HID_INOUT_DESCRIPTOR` macro is defined as:  
```c 
TUD_HID_INOUT_DESCRIPTOR(_itfnum, _stridx, _boot_protocol, _report_desc_len , _epin, _epsize, _ep_interval)
``` 
[image]

However, in the new_branch the macro now requires an additional parameter:  
```c
TUD_HID_INOUT_DESCRIPTOR(_itfnum, _stridx, _boot_protocol, _report_desc_len , **_epout**, _epin, _epsize, _ep_interval)
```

[image]

This results in a compilation error when switching branches once but I can't reproduce, parameter change can be reproduce everytime.

Environment  
- OS: Windows 11
- IDE: Void ( a fork of visual studio code )
- SDK Version: Latest stable version of Raspberry Pi Pico extension
- same TinyUSB submodule commit( the image below ): `86ad6e56c1700e85f1c5678607a762cfe3aa2f47` (verified via `git submodule status`)  
- same cmakelist.txt

[image]

I'm confused as to why this happens, and I haven't found a solution yet.

lurch, 2025-05-22:
> when working on my project but different branches of the same SDK version:

Could you please explain in more detail what this means? Currently the `master` branch of pico-sdk is at version [2.1.1](https://github.com/raspberrypi/pico-sdk/blob/master/pico_sdk_version.cmake) and the `develop` branch of pico-sdk is at version [2.1.2-develop](https://github.com/raspberrypi/pico-sdk/blob/develop/pico_sdk_version.cmake) so IMHO it doesn't really make sense to talk about "different branches of the same SDK version" 😕 

> same TinyUSB submodule commit

Just because each of your pico-sdk checkouts "wants" the same submodule commit of `lib/tinyusb` that doesn't necessarily mean that that's the actual commit of `tinyusb` that's checked out. Submodules can be confusing like that 😉 

So for each of your `pico-sdk` checkouts, please run each of these commands:
```
git status
git submodule status
git -C lib/tinyusb status
git -C lib/tinyusb log src/device/usbd.h
```

Note that if you've got multiple projects, with each of them using the same version of pico-sdk, you can have a standalone checkout of pico-sdk, and point each of your projects to that shared directory with the `PICO_SDK_PATH` variable (see https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf for some examples), rather than having each of your projects having its own copy of pico-sdk.

alt-0191, 2025-05-22:
> Could you please explain in more detail what this means? Currently the `master` branch of pico-sdk is at version [2.1.1](https://github.com/raspberrypi/pico-sdk/blob/master/pico_sdk_version.cmake) [...]

Apologies for the confusion — I misidentified the macro in my code (not an SDK issue). Thank you for your guidance, now I resolved it!

closed 2025-05-22 by alt-0191