raspberrypi/pico-sdk #2483: Bluetooth UUID

otacon6530, 2025-05-22:
Howdy,

Thank you for your time!

I just wanted to confirm if it is expected behavior that UUIDs are not retained when utilizing BTstack?

I am setting up my service with a UUID of B8E06067-62AD-41BA-9231-206AE80AB552, but the connected pico is ignoring it and sees it as 62AD41BA-9231-206A-E80A-B55100070001.

Steps to reproduce:
1. Create service
2. Start advertising

Second Pico:
1. Create Service
2. start blescanning
3. Verify connecting device by getAddressString()
4. Start discoverGATTServices();
5. get uuid via bleService->getUUID()->getUuidString()

[image]

my code: https://github.com/otacon6530/Area-51-Arcade-IR-Gun/tree/master/BTtest

Not sure if this is a BTstack issue or I am just doing it wrong or something, but let me know if you have any direction for me.

lurch, 2025-05-22:
> Let me know if there is another way to "nuke" it.

Which hardware are you using? Note that the flash_nuke that you linked to may not work as expected if you're using a 3rd-party board with more Flash memory than the official Pico and Pico 2 devices - see https://github.com/raspberrypi/pico-examples/issues/642#issuecomment-2842499419

otacon6530, 2025-05-22:
It is the raspberry pi pico w.

[image]

otacon6530, 2025-05-22:
I just noticed I posted this to the wrong git repository. I don't think this is an sdk issue. Rather an issue with the hardware library being used, so I am going to close it.

closed 2025-05-22 by otacon6530