raspberrypi/pico-sdk #2503: is it possible to control time between stop and start bits

kowshick16, 2025-06-05:
Here in below image, stop bit coming after 5ms in repeated start mode. is it possible to configure that time ? i want the stop bit to go immediately / the time i specified is sent.

i'm looking for ways to control this in both start-stop method and repeated start method ?

[image]

peterharperuk, 2025-06-05:
What code are you using? What hardware are you using?

kowshick16, 2025-06-06:
code for repeated start mode: 

uint8_t i2c_if_read(i2c_inst_t *i2c, uint8_t address, uint8_t *data, uint8_t size)
{
  return i2c_read_blocking(i2c, address, data, size, true);
}

kilograham, 2026-09-30:
`i2c_write_burst_blocking` (#1495) should cover this, so closing.

closed 2026-09-30 by kilograham