raspberrypi/pico-sdk #2507: How to build documentation?

marek22k, 2025-06-07:
Hello,

I would like to build the documentation as it can be found on https://www.raspberrypi.com/documentation/pico-sdk/index_doxygen.html for offline purposes. However, when I try I do not get the complete documentation. For example, the "High-Level SHA256 API" is missing, which is not in the documentation. Furthermore, the contrast of background and font is problematic. Hence the question: How to build the documentation? Is there a “pre-built” documentation?

I have tried the following:
```
$ git clone --recurse-submodules --branch master --depth 1 https://github.com/raspberrypi/pico-sdk.git
$ cmake -DCMAKE_BUILD_TYPE=Release -B builddir -S pico-sdk
$ cmake --build builddir --target docs
```

[image]

(see result in attachment)

[html.zip](https://github.com/user-attachments/files/20638330/html.zip)

kilograham, 2025-06-08:
in a fresh build directory, pass `-DPICO_PLATFORM=combined-docs` to CMake - you will only be able to build the `docs` target (other builds will fail)

marek22k, 2025-06-08:
That works. Thanks!

closed 2025-06-08 by marek22k

lurch, 2025-06-09:
This is all documented in Appendix B of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf 🙂