raspberrypi/pico-sdk #2523: Feature Request: Disable CYW43 debug output

negentropicdev, 2025-06-14:
I'm working on firmware for a device that can do host comms both over usb and over wifi as a tcp server.

The CYW43_WARN, _INFO, etc. periodically put stuff out on the usb (via their printf calls) that interferes with the usb communication.

There should be some defines that can disable that capability.

negentropicdev, 2025-06-14:
disregard, vscode initially sent me to a different file that didn't have the gating around the macros.

negentropicdev, 2025-06-14:
aaaand reopening because I'm unable to figure out a simple way to disable the outputs from build configuration.

kilograham, 2025-06-17:
see also #2253

lurch, 2025-06-18:
...and #2490

peterharperuk, 2025-06-18:
Those links are about bluetooth logging. It doesn't really address the fact that CYW43_PRINTF is a bit tricky to disable. 
We could add something to our cyw43_configport.h

```
#ifndef CYW43_PRINTF_ENABLED
#define CYW43_PRINTF_ENABLED 1
#endif

#if !CYW43_PRINTF_ENABLED
#define CYW43_PRINTF(...) (void)0
#endif
```

Then you just set CYW43_PRINTF_ENABLED=0 to disable all logging from the cyw43 code?

peterharperuk, 2025-06-19:
Yes, CYW43_LOGGING_ENABLED  sounds good

peterharperuk, 2025-07-21:
We settled on PICO_CYW43_LOGGING_ENABLED in the end. Set this to zero and  it should disable all printfs coming from cyw43.

commit b3aed37a (2025-07-14): Provide an easy way to disable cyw43 logging (#2535)

* Provide an easy way to disable cyw43 logging

All cyw43 logging uses CYW43_PRINTF by default. To disable logging you
have to define this to do nothing. There's no simple way to achieve
this. Make it easier by adding PICO_CYW43_LOGGING_ENABLED which can be
set to zero to disable CYW43_PRINTF.

Fixes #2523

* teensy style change

---------
closed 2025-07-21 by peterharperuk