raspberrypi/pico-sdk #2529: tools/check_board_header.py and maybe other tools

dhalbert, 2025-06-17:
tools/check_board_header.py gives syntax warnings for invalid escape sequences when used with Python 3.12 (at least). The `re` strings should be raw strings.

example:
```
/home/halbert/repos/adafruit/pico-sdk/tools/check_board_header.py:223: SyntaxWarning: invalid escape sequence '\s'
  if re.match("^\s*// For board detection", line):
/home/halbert/repos/adafruit/pico-sdk/tools/check_board_header.py:228: SyntaxWarning: invalid escape sequence '\s'
  m = re.match("^\s*// This header may be included by other board headers as \"(.+?)\"", line)
```

I'll submit a PR for this shortly

dhalbert, 2025-06-17:
Hmm, I searched for mentions of `check_board_header.py`, but somehow that PR didn't come up. I fixed more places than #2519 addressed, I believe. Maybe merge my #2530 first, and then #2519 should be mergeable, I think without conflicts.

will-v-pi, 2025-07-14:
The fixes have been merged into develop in #2519 and #2525 - the additional changes in #2530 are not required to fix those syntax errors

closed 2025-07-14 by will-v-pi