raspberrypi/pico-sdk #2534: `LLVM 20.1.0` change in `clang-runtimes` names

matsobdev, 2025-06-19:
Version `20.1.0` now treats `armv8m.main_soft_nofp` as `armv8m.main_soft_nofp_unaligned`, thus:
https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/cmake/preload/toolchains/pico_arm_cortex_m33_clang.cmake#L4
need to reflect that change. There are `armv6m_soft_nofp` and `armv6m_soft_nofp_unaligned` as well, but `diff` says they're the same.

kilograham, 2025-06-19:
sorry, can you give some more context. what is the effect of the Clang change. What does the fix do? What happens if you do nothing

kilograham, 2025-06-19:
"treats x as y" != "x has been renamed to y" :-) but ok, understand now, thanks

matsobdev, 2025-06-19:
Bad wording of mine, I'm still learning :D

PR #2542 (2025-06-30): Clang 20.1.0
It seems that
* [LLVM Embedded Toollchain for Arm](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm) has become [Arm Toolchain for Embedded](https://github.com/arm/arm-toolchain/blob/arm-software/arm-software/embedded/README.md)
* For unaligned-access safe targets, the defaut/only runtime now specifies `_unaligned` in the name

This PR adds the armv8m runtime `_unaligned` variant to our whitelist of runtimes to look for

fixes #2534