raspberrypi/pico-sdk #2550: I can't build in Visual Studio Code for the RPi PICO 2 W (RP2350 - Cortex)

user-redans, 2025-07-04:
Hi,

I recently bought a PICO 2 W and I want to build a simple blink example in VSCode using the Pico C/C++ SDK, but I'm getting the following compilation error:

Make Error at pico-sdk/cmake/pico_pre_load_platform.cmake:111 (message):
[cmake]   PICO_PLATFORM is specified to be 'rp2350', but PICO_BOARD='pico_w' uses
[cmake]   'rp2040' which is incompatible.  You need to delete the CMake cache or
[cmake]   build directory and reconfigure to proceed.  The best practice is to use
[cmake]   separate build directories for different platforms.

I've already set the following in CMakeLists.txt:

set(PICO_BOARD pico_w)
set(PICO_PLATFORM rp2350)

I don’t understand what I’m missing. Do you have any suggestions?

peterharperuk, 2025-07-04:
Pico_W uses rp2040. You want to set PICO_BOARD to pico2_w. It should work out PICO_PLATFORM from this so you shouldn't have to set it.

user-redans, 2025-07-06:
> Pico_W uses rp2040. You want to set PICO_BOARD to pico2_w. It should work out PICO_PLATFORM from this so you shouldn't have to set it.

Thanks for your answer! It works very well!

closed 2025-07-06 by user-redans