raspberrypi/pico-sdk #2553: missing CMakeLists.txt for cyw43-driver

webwarrior, 2025-07-07:
I ran the following command in a clean directory:

**git clone --recursive https://github.com/raspberrypi/pico-sdk.git**

I do not see the pico-sdk/lib/cyw43-driver/CMakeLists.txt created even after the following outputs from the command line:

Submodule path 'lib/btstack': checked out '501e6d2b86e6c92bfb9c390bcf55709938e25ac1'
Submodule path 'lib/cyw43-driver': checked out 'c1075d4bc440422cf2b2fd12c64a1f53f77660ee'
Submodule path 'lib/lwip': checked out '0a0452b2c39bdd91e252aef045c115f88f6ca773'
Submodule path 'lib/mbedtls': checked out '5a764e5555c64337ed17444410269ff21cb617b1'
Submodule path 'lib/tinyusb': checked out '86ad6e56c1700e85f1c5678607a762cfe3aa2f47'

**I then also tried the following in the pico-sdk directory, which did nothing..**.

git submodule update --init --recursive

I must be missing something here...

peterharperuk, 2025-07-07:
There is no such file so it won't be created

lurch, 2025-07-07:
What are you trying to do @webwarrior and/or what problem have you run into? Perhaps you're looking for https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_cyw43_driver/CMakeLists.txt ?