raspberrypi/pico-sdk #2556: Linker-generated veneer functions causing ISR_HardFault on RP2040

samyarsadat, 2025-07-13:
Yesterday, I was trying to test some firmware with FreeRTOS and micro-ROS on the RP2040.

After flashing the chip, I noticed that it wasn't doing anything (not registering a serial port over USB, not flashing its LED, etc.), so I decided to plug a debug probe in to see what was happening.

The chip was falling into an ISR HardFault before reaching the `main()` entrypoint. The hardfault was being caused by a linker-generated veneer function, specifically `__flash_do_cmd_veneer`, while trying to get the unique ID of the flash chip (`flash_get_unique_id()`) in the `runtime_init()` function.

I couldn't get more information about the type of HardFault, since Armv6-M cores don't implement the CFSR.

The firmware I was testing had a lot of new, untested code in it, so I thought I'd try a minimal FreeRTOS + micro-ROS example I had written and tested a few months ago, even though the hardfault wasn't directly related to any user-written code.

Even with my minimal example (which worked a few months ago), I still encountered the same hardfault as before. After this, I tried a minimal FreeRTOS example to see if maybe the hardfault was somehow being caused by the micro-ROS library. The minimal FreeRTOS example worked without any issues.

Minimal micro-ROS example: https://github.com/samyarsadat/Micro-ROS-RP2350/tree/rp2040-backport
Minimal FreeRTOS example: https://github.com/samyarsadat/Pico-Log-Library/tree/main/examples/freertos

This indicated that something was wrong with the micro-ROS library; however, I hadn't made any changes to the build configuration of micro-ROS since the last time I tested the example, and I didn't think that any changes to micro-ROS could've somehow interfered with a veneer function.

After some experimentation, I found that having even one micro-ROS RCL (ROS Client Library) function call (e.g., `rcl_get_default_allocator()`) in a function that never gets called results in a hardfault, but commenting that one line out resolves the hardfault.

```c
// main.c
#include <rcl/rcl.h>

rcl_allocator_t allocator;
void foo() {
    // Commenting this one line out resolves the hardfault.
    // It doesn't matter which micro-ROS function is called, it just has to be from the micro-ROS library.
    allocator = rcl_get_default_allocator();
}

int main() {
    return 0;
}
```

At this point, I decided to disassemble the binaries for both cases and compare the assembly code. The first thing I noticed was that `foo()` and `rcl_get_default_allocator()` were not included in the final assembly, which is expected as `foo()` isn't called anywhere. In fact, no micro-ROS-related functions appeared in either of the binaries. 

I then decided to inspect the assembly for `__flash_do_cmd_veneer`. This is where I noticed the main difference between the two binaries. All of the veneer functions in the failing binary appeared to be using shorter, "optimised" implementations compared to the ones in the other binary.

```asm
; for example, in the case of __flash_do_cmd_veneer:
__flash_do_cmd_veneer:
    ldr.w pc, [pc]
    .word 0x20000251

; vs.
__flash_do_cmd_veneer:
    push  {r0}
    ldr   r0, [pc, #8]
    mov   ip, r0
    pop   {r0}
    bx    ip
    nop
    .word 0x20000251

```

I have no prior assembly experience, but as far as I can tell based off of my research, the shorter implementation loads the address of the destination function directly into the program counter register, while the longer implementation uses `bx` to branch to the destination address. Directly loading an address into `pc` is apparently less safe from what I could find, and it appears to be the cause of the HardFault.

To verify that it indeed was the cause, I decided to try manually patch the failing binary to use `bx` instead of directly loading an address into `pc`.

```asm
; overwriting this:
__flash_do_cmd_veneer:
    ldr.w pc, [pc]
    .word 0x200002e9

; with this:
__flash_do_cmd_veneer:
    ldr r0, =0x200002e9
    bx  r0
; I understand that this doesn't preserve r0, but it worked fine for testing.
; preserving r0 would've resulted in the patch being larger than 8 bytes and not fitting.

```

After the patch, the program managed to reach `main()` without a hardfault.

I'm not really sure why the linker chooses the shorter implementation when any micro-ROS function calls are present. I tried getting the linker to generate the shorter veneers without the micro-ROS library present at all, but I couldn't get it to do it. 

Though, forcing the linker to generate position-independent veneers by setting `--pic-veneer` in the linker options in the CMake config of the project solves the issue and makes the linker always generate the longer implementation.

I'm not sure whether this is a bug or intended behaviour, but it seems strange to me that it is possible for the linker to generate veneers that cause hardfaults. This is the first time I'm filing a bug report like this, so I apologise for any mistakes I might've made.

Just as a side note, I am using the latest version of the Pico SDK from the master branch, with `arm-none-eabi-gcc` version 13.2.1 and `arm-none-eabi-ld` version 2.42.

lurch, 2025-07-13:
Wow, that's an incredibly deep-dive level of debugging, especially for a 16-year-old, well done! 👏 

I'm afraid that you've already gone beyond my low-level knowledge; but as an extra debugging step, before the much more knowledgeable @kilograham offers his insight, would you mind seeing if the latest `develop` branch of Pico SDK produces the same behaviour please?

samyarsadat, 2025-07-13:
> Wow, that's an incredibly deep-dive level of debugging, especially for a 16-year-old, well done! 👏

Thank you very much!

> ...would you mind seeing if the latest develop branch of Pico SDK produces the same behaviour please?

I just tried that now, same results I'm afraid.

> I looked up the [BX instruction](https://developer.arm.com/documentation/dui0379/e/arm-and-thumb-instructions/bx?lang=en) and see that it says "The BX instruction causes a branch to the address contained in Rm and exchanges the instruction set, if required", so it's easy to imagine that if the CPU is in Thumb mode and you suddenly jump directly into ARM code, or vice-versa, that it'd cause a HardFault.

I also did notice that when looking up `bx`, but I didn't check with the debugger because I thought that directly loading a value into `pc` couldn't switch the processor out of Thumb mode. The [ARM documentation](https://developer.arm.com/documentation/dui0662/b/The-Cortex-M0--Processor/Programmers-model/Core-registers?lang=en) for the Cortex-M0+ states that:
> ...Attempts to write the EPSR using the MSR instruction are ignored...
> ...The following can clear the T bit to 0:
> - Instructions BLX, BX and POP{PC}.
> - Restoration from the stacked xPSR value on an exception return.
> - Bit[0] of the vector value on an exception entry.

According to this, loading a value into `pc` shouldn't be able to clear the Thumb state bit, and so shouldn't be able to switch the processor out of Thumb mode, which makes sense given that Cortex-M0+ is Thumb-only.

Having said that, I ckecked with the debug probe, and bit 24 of the `PSR` (T bit of the `EPSR`) appears to be 0 on the stack frame for `__flash_do_cmd_veneer`. I'm a bit confused as to how the processor got switched out of Thumb mode.

[image]
<br><br>

Maybe this is a bug with the debugger and the cause of the hardfault is something else, I'm not really sure. Regardless, somethng about loading the address straight into `pc` is causing a hardfault.

kilograham, 2025-07-13:
How are you compiling this? `ldr.w pc, [pc]` is not a valid Cortex M0+ instruction, which would seem to imply you are compiling with the wrong target. Is it possible you re-used an existing build directory which you had been playing with RP2350, or are you in fact using RP2350 and not RP2040 as stated.

kilograham, 2025-07-13:
Note that the veneers are added by the linker, so are you for some reason using a customer linker?

samyarsadat, 2025-07-13:
> How are you compiling this? `ldr.w pc, [pc]` is not a valid Cortex M0+ instruction, which would seem to imply you are compiling with the wrong target. Is it possible you re-used an existing build [...]

> Note that the veneers are added by the linker, so are you for some reason using a customer linker?

My target is set correctly in the CMake configuration for the Pico SDK project, I am not using a custom linker, I did make sure to do clean build, and I am using the RP2040 for certain. 

It turns out, however, that I had accidentally overwritten one line in the toolchain configuration for the micro-ROS library build while merging some code from the RP2350 branch of the same repo... So instead of having `-mcpu=cortex-m0plus` in the compilation flags, I had `-mcpu=cortex-m33`. I presume this is what was causing the linker to generate invalid Cortex-M0+ assembly when linking against the micro-ROS library.

I still had `set(CMAKE_SYSTEM_PROCESSOR cortex-m0plus)` at the top of the file, so I didn't think to check further down. It probably worked a few months ago due to some build caching somewhere from before I overwrote the file.

I feel incredibly stupid now, sorry for the trouble. Thank you for the help. I will close the issue now.
Though, I'm surprised the linker didn't produce a warning because of the mismatch.

samyarsadat, 2025-07-13:
Although, may I ask why forcing the linker to generate position-independent veneers (by setting `--pic-veneer`) casued it to generate valid Cortex-M0+ assembly for the veneers? Is it because the other implementation isn't considered PIC-compliant? Just out of curiosity.

closed 2025-07-13 by samyarsadat