raspberrypi/pico-sdk #2572: 2.1.1 Tag killed pipelines over night

anobit, 2025-07-18:
Two weeks ago a oversight regarding the picotool dependency was fixed:
https://github.com/raspberrypi/pico-sdk/commit/9a4113fbbae65ee82d8cd6537963bc3d3b14bcca

This in turn caused CI pipelines to fail since today:
https://github.com/raspberrypi/picotool/issues/256

Since the 2.1.1 Tag does not reflect the changes in the before mentioned commit, even rebuilding docker images does not help.

While this may not be the go-to solution for a problem hinting at wrongly setup CI builds, the 2.1.1 Tag should contain the before mentioned commit.

kilograham, 2025-07-18:
yeah, we can't modify existing tags... we could add a new interim tag

will-v-pi, 2025-07-18:
Yes, auto downloading & building picotool is broken on the 2.1.0 and 2.1.1 SDK tags, due to that issue. A new interim tag would probably be a good idea.

The best fix is to build & install picotool separately in your pipeline - see the picotool readme for instructions, or see the picotool CI jobs for demonstrations of how to do it.