raspberrypi/pico-sdk #2581: Add more defines related to partition tables

will-v-pi, 2025-07-22:
It would be good to add some of the defines from https://github.com/raspberrypi/pico-examples/pull/627#pullrequestreview-3042818651 into the SDK, including adding defines for the family names (which can also be used by picotool).

kilograham, 2025-07-22:
punting this to 2.3.0 as we should probably add a library related to all this

kilograham, 2025-07-23:
yeah, i think it is fine - it shows you how to do it for now

andygpz11, 2025-07-30:
Hi @will-v-pi As a part of this work, would it be possible to add a size definition for the `workarea` required when using the bootrom's flash API please?

We can then replace some hardcoded instances.

Thanks.

andygpz11, 2025-07-30:
Whilst I'm here, is there any appetite for adding something like:
 `#define UF2_BLOCK_SIZE   512`    ?

I realise there's a `static_assert(...)` in `uf2.h` to guard against this struct being sized _other_ than 512 but it would be nice if source could use a common definition - this must be immutable by now.
Apologies if I've missed something I could use.
Thanks.

andygpz11, 2025-07-31:
@lurch Thanks.  Yes, "... that was not the droid I was looking for" :)