raspberrypi/pico-sdk #2590: When using clangd, c++ is broken

StiglCZ, 2025-07-24:
When I use C with pico-sdk, it works just fine, I just use:
`set(CMAKE_EXPORT_COMPILE_COMMANDS ON)`
in my CMakeLists.txt
And 
```
Diagnostics:
  Suppress:
    - pp_file_not_found_angled_include_not_fatal
    - unused-includes
CompileFlags:
  Add:
    -I/usr/arm-none-eabi/include
```
in my .clangd, and that works great.
When I however generate the project with C++(using rpi-pico-project-generator), it says:
`In included file: "No thread API" [pp_hash_error]`
And various other errors including pp_file_not found
How would I make clangd accept it with c++ correctly? I can suppress these errors, but then it doesnt seem like clangd does autocomplete at all.

kilograham, 2025-07-25:
Questions better asked on the forums https://forums.raspberrypi.com/viewforum.php?f=143

closed 2025-07-25 by kilograham

lurch, 2025-07-28:
> (using rpi-pico-project-generator)

Are you talking about https://github.com/raspberrypi/pico-project-generator ? I think that's no longer supported, and has been replaced by https://github.com/raspberrypi/pico-vscode

StiglCZ, 2025-07-28:
I will propably still keep using it since I dont use vscode frankly.