raspberrypi/pico-sdk #2611: Pico-slave library does not respond to NAK

CMakeMain, 2025-07-29:
Hi, I am currently trying to communicate with my RP2040 with my ESP32. I am sending a test read with a size of 1 byte. According to the I2C spec, the last byte should be NAKed but the Pico slave library doesn't seem to accept the NAK and stop sending data.

Here is my slave ISR:
```
void i2c1_irq_handler(i2c_inst_t *i2c, i2c_slave_event_t event){
    switch(event){
        case I2C_SLAVE_RECEIVE:
            rx_buffer[pointer_rx] = i2c_read_byte_raw(i2c);
            pointer_rx = (pointer_rx + 1) % RX_BUFFER_MAX_SIZE;
            break;
        case I2C_SLAVE_REQUEST:
            if(size_tx > 0){
                i2c_write_byte_raw(i2c, tx_buffer[pointer_tx]);
                pointer_tx = (pointer_tx + 1) % TX_BUFFER_MAX_SIZE;
                size_tx--;
            }
            break;
        default:
            break;
    }
}
```

Here is my main function:
```
int main(){
    ....
    gpio_init(I2C_SDA);
    gpio_set_function(I2C_SDA, GPIO_FUNC_I2C);
    gpio_pull_up(I2C_SDA);
    gpio_init(I2C_SCL);
    gpio_set_function(I2C_SCL, GPIO_FUNC_I2C);
    gpio_pull_up(I2C_SCL);
    i2c_init(i2c1, 400 * 1000); // 400000 khz
    i2c_slave_init(i2c1, SLAVE_ADDR, &i2c1_irq_handler);

    ...
}
```

CMakeMain, 2025-07-30:
Hello, has there been any update to this situation?

CMakeMain, 2025-07-31:
Update: I have just confirmed this SCL stuck low also occurs when using the function ```i2c_set_slave_mode()```

CMakeMain, 2025-07-31:
I am not sure why that matters when there is potentially some error in the actual HAL, but here goes nothing: Steps to reproduce: Copy the ISR above, intialise RP2040 with slave mode. Create TX buffer with a size >0, and populate with arbitrary data. Then attempt to read and view the logic analyser.

JamesH65, 2025-08-01:
Did you fix the issue?

CMakeMain, 2025-08-01:
No, I haven't. There is something clearly wrong with the pico sdk and it has just lead me through an endless rabbit hole of errors. Firstly, the I2C_SLAVE_INIT function will hang the SCL line indefinitely. Then, for some reason when I set it to slave mode first then initialise the I2C peripheral it will work? But then when I woke up today it doesn't work again. I have given up in using the RP2040. I will probably use something with a more reliable SDK for future school projects.

JamesH65, 2025-08-01:
It seems unlikely it's an SDK issue, a lot of people/companies use I2C with the RP2040 with no problems. That said, there is possibly an edge case in here that has reared its head. When engineering time allows we'll take a look. It would be worth updating to the latest SDK (released in the last two days) just in case it's an old bug that has been fixed. 

If the examples are not working, then it's almost certainly an issue at the implementation end. We know the examples work when the HW is working and everything is plugged in correctly. This is emphasised by it working one day but not the next - the SW hasn't changed in that time, which can indicate HW right on the edge of working.

lurch, 2025-08-01:
> Although in the end the `pico_i2c_slave` API wasn't powerful enough, I needed more control over the configuration of the peripheral (eg 7-bit vs 10-bit addressing, FIFO depth), and needed to handle more of the IRQs. So I used direct register access to control it (which was actually pretty easy, the I2C peripheral itself is very good).

@dpgeorge Please feel free to create an issue (or a PR!) requesting/adding the parts that you feel are missing in the SDK's current I<sup>2</sup>C API 🙂