raspberrypi/pico-sdk #2616: pio.c needs 32u to be changed to NUM_BANK0_GPIOS on line 347 to support RP2350B

peterk268, 2025-08-07:
Findings found here:
https://picopalgbc.wordpress.com/2025/08/06/97-swap-pclk-and-den-port-code-to-pico-pal/

Line 347 in src/rp2_common/hardware_pio/pio.c currently uses 32u as the check for the gpio pin number threshold but this doesn't support the RP2350B with 48 GPIO so this would return the error code -5 or invalid arg from this line. 

`    invalid_params_if_and_return(PIO, pin >= 32u, PICO_ERROR_INVALID_ARG);
`

Thus the fixed line is here:
`    invalid_params_if_and_return(PIO, pin >= NUM_BANK0_GPIOS, PICO_ERROR_INVALID_ARG);
`

peterk268, 2025-08-07:
Ah that's my bad. I guess once I kept going with fixing the error codes and then I set the base to 16 after making these changes it lead me to believe that this was an issue.

lurch, 2025-08-07:
Can this `pico-sdk` issue be closed now then @peterk268 ?

closed 2025-08-07 by lurch