raspberrypi/pico-sdk #2623: Improving HID throughput consistency on Pico W

mitchellcairns, 2025-08-09:
I’m working on a Bluetooth Classic HID device on the Raspberry Pi Pico W (RP2040 + CYW43), using BTstack from the Pico SDK. I’m trying to achieve a consistent reporting interval for gamepad input.

My btstack_config.h:
```c
#ifndef _PICO_BTSTACK_BTSTACK_CONFIG_H
#define _PICO_BTSTACK_BTSTACK_CONFIG_H

// BTstack features that can be enabled
#define ENABLE_LE_PERIPHERAL
#define ENABLE_LE_CENTRAL
#define ENABLE_L2CAP_LE_CREDIT_BASED_FLOW_CONTROL_MODE
//#define ENABLE_LOG_INFO 
#define ENABLE_LOG_ERROR
#define ENABLE_PRINTF_HEXDUMP
#define ENABLE_SCO_OVER_HCI

// BTstack configuration. buffers, sizes, ...
#define HCI_OUTGOING_PRE_BUFFER_SIZE 4
#define HCI_ACL_PAYLOAD_SIZE (69)
#define HCI_ACL_CHUNK_SIZE_ALIGNMENT 4

#define MAX_NR_L2CAP_CHANNELS 2
#define MAX_NR_L2CAP_SERVICES 1
#define MAX_NR_SERVICE_RECORD_ITEMS 4
#define MAX_NR_BTSTACK_LINK_KEY_DB_MEMORY_ENTRIES 3

#define MAX_NR_SM_LOOKUP_ENTRIES 3
#define MAX_NR_WHITELIST_ENTRIES 16
#define MAX_NR_LE_DEVICE_DB_ENTRIES 16

#define MAX_NR_AVDTP_CONNECTIONS 0
#define MAX_NR_AVDTP_STREAM_ENDPOINTS 0
#define MAX_NR_AVRCP_CONNECTIONS 0
#define MAX_NR_BNEP_CHANNELS 0
#define MAX_NR_BNEP_SERVICES 0

#define MAX_NR_GATT_CLIENTS 0
#define MAX_NR_HCI_CONNECTIONS 1
#define MAX_NR_HID_HOST_CONNECTIONS 1
#define MAX_NR_HIDS_CLIENTS 0
#define MAX_NR_HFP_CONNECTIONS 0

#define MAX_NR_RFCOMM_CHANNELS 0
#define MAX_NR_RFCOMM_MULTIPLEXERS 0
#define MAX_NR_RFCOMM_SERVICES 0

// Limit number of ACL/SCO Buffer to use by stack to avoid cyw43 shared bus
// overrun
#define MAX_NR_CONTROLLER_ACL_BUFFERS 8
#define MAX_NR_CONTROLLER_SCO_PACKETS 0

// Enable and configure HCI Controller to Host Flow Control to avoid cyw43
// shared bus overrun
#define ENABLE_HCI_CONTROLLER_TO_HOST_FLOW_CONTROL
#define HCI_HOST_ACL_PACKET_LEN HCI_ACL_PAYLOAD_SIZE
#define HCI_HOST_ACL_PACKET_NUM 8

#define HCI_HOST_SCO_PACKET_LEN 0
#define HCI_HOST_SCO_PACKET_NUM 0

// Link Key DB and LE Device DB using TLV on top of Flash Sector interface
#define NVM_NUM_DEVICE_DB_ENTRIES 16
#define NVM_NUM_LINK_KEYS 16

// We don't give btstack a malloc, so use a fixed-size ATT DB.
#define MAX_ATT_DB_SIZE 512

// BTstack HAL configuration
#define HAVE_EMBEDDED_TIME_MS

// map btstack_assert onto Pico SDK assert()
#define HAVE_ASSERT

// Some USB dongles take longer to respond to HCI reset (e.g. BCM20702A).
#define HCI_RESET_RESEND_TIMEOUT_MS 1000

#define ENABLE_SOFTWARE_AES128
#define ENABLE_MICRO_ECC_FOR_LE_SECURE_CONNECTIONS

#define HAVE_BTSTACK_STDIN

#ifdef ENABLE_CLASSIC
//#define ENABLE_L2CAP_ENHANCED_RETRANSMISSION_MODE
#endif
#endif  // MICROPY_INCLUDED_EXTMOD_BTSTACK_BTSTACK_CONFIG_H
```

## Current behavior
HID report size: 64 bytes (plus 1-byte HIDP header, plus 4-byte L2CAP header = 69 bytes total ACL payload)

At a 4 ms target: ~95% of reports arrive on time, ~5% deviate (up to 20 ms delay)
At 2 ms target: ~60% on time, ~40% deviate (same jitter pattern)

## Question
Is there an ideal buffer/flow-control configuration for the Pico W/CYW43 that would yield more consistent throughput for small, frequent HID interrupt packets?

Are there recommended configuration values for high-rate HID traffic?

Any best practices for tuning HCI_ACL_PAYLOAD_SIZE relative to HID packet size?

Are there known limits with the CYW43 transport that would make sub-4 ms intervals inherently unstable?

## Additional notes
Reports are only sent when BTstack’s L2CAP_EVENT_CAN_SEND_NOW fires (no busy-looping or direct l2cap_send). 

I understand the model where the callback event can be fired to indicate when the RP2040 is allowed to fire another input report, however it would be nice to have some sort of flag, for unreliable packets especially to just 'send' a packet without a need to worry about its status etc. This seems to add unwanted latency for these sorts of use-cases, where we are simply interested in sending a constant stream of data at a fixed interval. 

Thanks for your time!

mitchellcairns, 2025-08-10:
I did put up a google group discussion in the official BTStack group as well, to basically help anyone who searches for such an issue in the future to find something relevant: https://groups.google.com/g/btstack-dev/c/p1329YxzAH8/m/LmmGg-iWBgAJ

peterharperuk, 2025-08-11:
I don't have any useful answers to your questions. What async method are you using? Polling, background thread safe or Freertos? I would check when the irq goes off cyw43_gpio_irq_handler. Is it that or cyw43_poll_func that's delayed. Does cyw43_ll_bt_has_work return true when expected? hci_transport_cyw43_process may also be relevant. I'd instrument these functions with a logic analyzer and see if there's any correlation with the delays you're seeing.

mitchellcairns, 2025-08-11:
To clarify, I’m not fully familiar with all the internal interactions between cyw43_gpio_irq_handler, cyw43_poll_func, and the other functions you mentioned. I do not currently have access to tools to probe the Pico-W with a logic analyzer during all of this. 

To simplify things and rule out layering/timing issues, I’m currently not using FreeRTOS or any complex task scheduling—just basic polling and background-thread-safe driver modes.

I put together a minimal test repository to demonstrate the timing inconsistencies with both polling and background-thread-safe approaches:

https://github.com/mitchellcairns/hid-test-picow

I’m pairing this with [this Windows tool](https://github.com/cakama3a/Polling/releases/tag/1.3.1.1) to verify the effective polling rate on the host side.

From my observations, it doesn’t seem to matter which async method I use; when using the BTstack timers as recommended, I get quite variable latency on when my HID report gets serviced.

The approach that appears “best” still suffers from interruptions causing inconsistent intervals. Granted this requires going outside of the recommended approach of using a timer and having BTStack automatically handle that data source.  Also this seems to still work, despite never requesting a 'can send now' event. 

Is there any path here to have a more consistent report interval? 

Thanks again for your time!