raspberrypi/pico-sdk #2625: Pico 2 W Bluetooth LE Throughput

JW5412, 2025-08-13:
I'm working on a project that needs to maximize the BLE throughput from the Pico 2 W to an iOS device.
Per the "Accessory Design Guidelines for Apple Devices" the connection interval is limited to 15ms, which is what I'm connecting at even if I request 7.5ms. Latency is set to zero.

I've defined ENABLE_LE_DATA_LENGTH_EXTENSION and this bumps my usable payload up to 249 bytes per packet.

Since both ends of my link support BT5.0, I've tried to set the PHY to use 2Mbps modulation, with this command.
gap_le_set_phy(0x01, 0, 0x2, 0x2, 0);
But I don't think it's being respected. I couldn't find an example where this API is used.

As I understand it the BT stack should be able to send multiple packets per connection event (up to 7?).

At 1 packet of 249 bytes per 15ms I should get ~16.6kBps, but I'm only seeing ~5KBps come through the ATT.

Has anyone had success in maximizing the BLE throughput?

dpgeorge, 2025-08-14:
As a point of reference: in MicroPython we have BLE performance tests and for a Pico 2 W communicating with another Pico 2 W, we can achieve up to about 36500 bytes/second using GATT write-without-response transfers.

See https://github.com/micropython/micropython/blob/master/tests/multi_bluetooth/perf_gatt_char_write.py

There's nothing special going on, it uses an MTU of 131, a connection interval request between 10ms and 30ms, and a latency of 4.

matsobdev, 2025-08-15:
I believe DLE was fixed fixed with SDK version 2.2.0 with corresponding CYW43 driver (containing fixed firmware). Double check if you use one.

JW5412, 2025-08-16:
@matsobdev I am on the 2.2.0 SDK and the BTFW reports back as "CYW4343A2_001.003.016.0031.0000_Generic_SDIO_37MHz_wlbga_BU_dl_signed" which aligns with the latest cyw43-driver/firmware/cyw43_btfw_43439.h

Near as I can tell DLE is working, I can send 234 byte payloads over the air (DLE max is 249).
But it doesn't seem that I can sustain that throughput at each connection interval (15ms), let alone send multiple packets per connection interval to an iOS central.

Not sure if anyone has gotten BTLE 2Mbps PHY to work.
gap_le_set_phy(con_handle, 0, 0x2, 0x2, 0);
gap_set_connection_phys(0x2)

@dpgeorge Thanks this is a helpful datapoint! I need to see if I can adapt your MicroPython reference to work with an iOS central. This aligns with a PunchThrough blog that says with DLE ~50KBps should be achievable.

@ricardoquesada @mringwal Were you ever successful in getting more than one packet per connection interval?
I saw that you had a recommendation to stop scanning. Are these the correct APIs to call post connection?

gap_connectable_control(false);
gap_advertisements_enable(false);
gap_stop_scan();

Thanks!

mringwal, 2025-08-16:
@JW5412 The CYW43439 does not support 2M-PHY.

If you run gatt_streamer and connect from an iOS device with a GATT browser and enable notifications for one of the characteristics, you should get data at 35 kB/s without DLE and about 50 kB/s with DLE. (And yes, this means more than one packet per connection interval).

If you don't enable anything Classic, you don't need to disable it. Connectable control will be enabled automatically if there's an L2CAP service registered and Discoverable control is always manual anyways. Your LE Peripheral should not scan either and without explicitly enabling it BTstack will not send advertisements if it's already connected via LE.

--

Another line of attack would be to use GATT over Classic (does not need MFi). When Classic + LE are enabled and you pair over LE, the devices will perform Cross-transport key derivation which will allow you to create a Classic connection to iOS if you setup the GATT SDP records - see gatt_streamer_server as an example. iOS will then use Classic for GATT which can give you higher throughput. (It's not LE, but hey, depends on your goal).

closed 2025-08-17 by JW5412