raspberrypi/pico-sdk #2626: Compile error with PICO_DISABLE_SHARED_IRQ_HANDLERS

jepler, 2025-08-13:
CMakeLists.txt:
```cmake
cmake_minimum_required(VERSION 3.13)

include(pico_sdk_import.cmake)

project(test_project C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
pico_sdk_init()

add_executable(main
  main.c
)

pico_enable_stdio_usb(main 1)
pico_enable_stdio_uart(main 1)

pico_add_extra_outputs(main)

target_link_libraries(main pico_stdlib)

target_compile_definitions(main PRIVATE PICO_DISABLE_SHARED_IRQ_HANDLERS=1)
```

```
/**
 * Copyright (c) 2020 Raspberry Pi (Trading) Ltd.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#include "pico/stdlib.h"

int main() {
    const uint LED_PIN = PICO_DEFAULT_LED_PIN;
    gpio_init (LED_PIN);
    gpio_set_dir (LED_PIN, GPIO_OUT);
    while (true) {
        gpio_put (LED_PIN, 1);
        sleep_ms (250);
        gpio_put (LED_PIN, 0);
        sleep_ms (250);
    }
}
```

Results of building:
```
$ cmake -S . -B build -DPICO_SDK_PATH=$HOME/src/pico-sdk && make -C build -j$(nproc)
[…]
/home/jepler/src/pico-sdk/src/rp2_common/hardware_irq/irq.c: In function 'irq_set_exclusive_handler':
/home/jepler/src/pico-sdk/src/rp2_common/hardware_irq/irq.c:238:5: error: implicit declaration of function 'set_raw_irq_handler_and_unlock' [-Wimplicit-function-declaration]
  238 |     set_raw_irq_handler_and_unlock(num, handler, save);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

This appears to be a regression since 2.1.1 and affects 2.2.0 and develop.

Moving the definition of `static void set_raw_irq_handler_and_unlock` so it is not conditional on ` #if !PICO_DISABLE_SHARED_IRQ_HANDLERS && !PICO_NO_RAM_VECTOR_TABLE`. However I do not know whether this change is correct. I can submit it as a PR if desired.

jepler, 2025-08-13:
This affects building rp2040-doom 😜

kilograham, 2025-08-13:
> This affects building rp2040-doom 😜

NOOOOOOOOO!

Seriously tho I did build it with 2.2.0 before release so will have to see what happened - I think I was using the rp2350 branch

PR #2631 (2025-08-19): fix regression caused by moving set_raw_irq_handler_and_unlock()
fixes #2626
closed 2025-08-19 by lurch