raspberrypi/pico-sdk #2633: Investigate TLS incompatibility with mbedtls 3

peterharperuk, 2025-08-20:
I have some code that connects to a raspberry pi server over TLS. It's failing with the latest SDK which uses mbedtls 3.x

```
/home/peterh/source/pico-sdk1/lib/mbedtls/library/ssl_tls12_client.c:2438: mbedtls_pk_verify() returned -19968 (-0x4e00)

mbedtls_ssl_handshake failed: -19968
Request completed with result 4 len 0 server_response 0 err -15 in 556ms
Request failed

```

If I change back to mbedtls 2.x the TLS handshake works ok. Investigate what's changed in mbedtls to cause this.

```
/** The signature is not valid. */
#define MBEDTLS_ERR_ECP_VERIFY_FAILED                     -0x4E00

```

peterharperuk, 2025-08-20:
it comes from here...

```
Thread 1 "rp2350.cm0" hit Breakpoint 3, mbedtls_ecdsa_verify_restartable (grp=0x20015688,
    buf=0x20081b10 "w\266\264Ȃל\350l?\233S!\231\206\206\331\307I\021\r[\266", blen=32, Q=0x200156f0,
    r=0x20081a68, s=0x20081a60, rs_ctx=0x0) at /home/peterh/source/pico-sdk1/lib/mbedtls/library/ecdsa.c:579
579             ret = MBEDTLS_ERR_ECP_VERIFY_FAILED;
```

will-v-pi, 2025-08-25:
Looks like you spotted this in pico-examples before - https://github.com/raspberrypi/pico-examples/issues/599

croepha, 2026-02-25:
So, when I realized that the pico doesn't have an RTC, I kinda gave up on TLS, I was planning on using libsodium with a simpler protocol 

EDIT: ok, so it might have an RTC, but its not like it will keep time across a reset

amitnjha, 2026-03-06:
> Hello, [@croepha](https://github.com/croepha) could you pls post your mbedtls_config.h file because I am facing the same problem. As soon as I add MBEDTLS_ALLOW_PRIVATE_ACCESS to mbedtls_config.h I get the error message "No mbedtls_ms_time available". When I then add MBEDTLS_PLATFORM_MS_TIME_ALT I get "MBEDTLS_PLATFORM_MS_TIME_ALT defined but not all prerequisites". So I've ran out of ideas. Thx

@Florian-Wilhelm Were you able to figure this out? I am running into same issue.