raspberrypi/pico-sdk #2640: Add Raspberry Pi 500's RP2350 board support

jhordies, 2025-09-01:
The Raspberry Pi 500 contains an embedded RP2350 microcontroller that handles keyboard input and can be reprogrammed for custom applications. This issue tracks adding official board support to enable developers to create custom firmware for the Pi 500's RP2350 chip.

**Hardware Details:**

Microcontroller: RP2350A (ARM Cortex-M33 + RISC-V Hazard3)
Flash: W25X10CL (DSPI mode, not QSPI like standard Pico 2)
USB: Connected to Pi 500 USB lines, functions independently
GPIO: All pins reserved for 8×18 keyboard matrix and system functions
System pins: GP17 (LED), GP19 (power control), GP20 (power key), GP25 (Caps LED)
Power: Independent power, can operate when main Pi is off

**Use Cases:**

Custom USB devices (MIDI controllers, HID devices, CDC serial)
Keyboard firmware modifications and enhancements
Educational embedded systems projects (USB-focused)
Debug and development tools via UART (GP16)

**Implementation Requirements:**

Board definition with W25X10CL flash support (DSPI mode)
Complete GPIO pin mapping and system function documentation
USB configuration for standalone operation
Compatibility with existing Pi 500 hardware

Note: Unlike standard Pico boards, the Pi 500 has no GPIO pins available for general use - all are reserved for keyboard matrix and system functions but are accessible via the keyboard's FFC ribbon. 

**⚠️ CRITICAL WARNING**: Custom firmware must implement power button handling (GP19 control) or the Pi 500 power button will stop working, potentially making the device unbootable. Always include power management code from the original QMK firmware.

lurch, 2025-09-01:
See also https://github.com/raspberrypi/QMK/issues/4
And also https://youtu.be/omYWRb1dLA4?t=684 😉

jhordies, 2025-09-01:
If this PR https://github.com/raspberrypi/pico-sdk/pull/2641 gets merged, I shall open a PR for:

A tinyusb based firmware replacement: 
https://github.com/jhordies/tinyusb/tree/add-pi500-keyboard-example
or
A tinyusb based firmware replacement + midi key board (switch by pressing Shift L + Shift R for 500ms)
(white keys are on the qwer... line, the black keys are the number keys)
https://github.com/jhordies/tinyusb/tree/add-pi500-dual-mode-example

lurch, 2025-09-03:
I've left you some more feedback on your PR too.

peterharperuk, 2025-09-03:
I don't see why this needs to be in the SDK at all.

jhordies, 2025-09-03:
> > Can we agree on the key matrix scanning not be the exclusive usage of the FCC ribbon connector and have the SDK act as much as it would be if it was a standard pico ?
> 
> Sorry, but no.
> 
> > I know the pi500 is not marketed as a development board but I wouldn't restrict the application scope to solely key matrix scanning of the physical keyboard.
> 
> I don't think that's something that we'd be willing to merge into pico-sdk (but of course you're free to do that in your personal header-files). IMHO this would just be too much of a niche use-case (and also risks people damaging the FCC connector on their Pi 500).

Let me quote @geerlingguy : 
_The next question, since it seems obvious the Pi 500's PCB is intended for something more—could we see an enthusiast-grade mechanical keyboard version of the Pi 500, with all the bells and whistles? I've already been asking System76 if they might consider designing bottom plate for the Launch keyboard that adapts the Pi 500..._
source : [https://www.jeffgeerling.com/blog/2024/raspberry-pi-500-uses-qmk-firmware-built-keyboard](url)

The Pi 500 itself is a niche product - a keyboard computer. Adding SDK support actually broadens its market appeal by enabling:
Custom keyboard firmware (mechanical keyboard enthusiasts), Educational embedded systems projects, MIDI controllers and creative applications,..

The RP2040 chip is fully capable and I believe shouldn't be artificially limited. Users who want to experiment with custom firmware should have proper SDK support rather than reverse-engineering or using unofficial definitions.

When developing a usb gadget device (eg:midi), the first thing one need is to connect it via usb on a computer (eg: sound module), the pi500 has this built in, that'd be a pity not to be able to leverage it while the hardware supports it.

About the risk of damaging the flat cable, raspberry pi have been using Flat cables as a standard way for user handled extensions

[image]

and an even more fragile FPC for pcie lane on the pi5

[image]

> I don't see why this needs to be in the SDK at all.

It enabled me to create this custom firmware that is both a computer keyboard and a piano midi keyboard without hardware modification:
https://github.com/jhordies/tinyusb/tree/add-pi500-dual-mode-example/examples/device/pi500_dual_mode
Which I think is a nice educational add-on for pi 500 owners. 

It was fun and useful to me, that's why I thought I would spend the extra effort to share it with the maker community, I'm sorry if this was not the appropriate place to do so.

Thanks @lurch for your review efforts

lurch, 2025-09-04:
> It enabled me to create this custom firmware that is both a computer keyboard and a piano midi keyboard **without hardware modification**:

(emphasis added by me)

I personally have no problem with that. It's the suggestion of connecting "other hardware" to the keyboard FPC connector that I have a problem with. There's a **big** difference between damaging a [Raspberry Pi Pico](https://www.raspberrypi.com/products/raspberry-pi-pico/) which costs about £4, and damaging a [Raspberry Pi 500](https://www.raspberrypi.com/products/raspberry-pi-500/) which costs about £85.