raspberrypi/pico-sdk #2666: Incorrect behavior for add_program_at_offset for pins > 31

drewpo28, 2025-09-15:
Hello,

I guess the change in method 'add_program_at_offset' need to be done for board 2350 in case of used pins > 31 with `pio_set_gpio_base(pio, 16);`

Please remove the line - `static_assert(PIO_GPIOBASE_BITS == 16, ""); // only works for gpio base being 0 or 16`

```
static int add_program_at_offset(PIO pio, const pio_program_t *program, uint offset) {
    int rc = add_program_at_offset_check(pio, program, offset);
    if (rc != 0) return rc;
    for (uint i = 0; i < program->length; ++i) {
        uint16_t instr = program->instructions[i];
#if PICO_PIO_USE_GPIO_BASE
        if (pio_instr_bits_wait == _pio_major_instr_bits(instr) && !((_pio_arg1(instr) & 3u))) {
            // wait GPIO will include only the 5 lower bits of the GPIO number, so if the GPIO
            // base is 16 we need to flip bit 4 (which is equivalent to subtracting 16 from
            // the original number 16-47 stored as 16-31 and 0-15)
            static_assert(PIO_GPIOBASE_BITS == 16, ""); // only works for gpio base being 0 or 16
            instr ^= (uint16_t)pio_get_gpio_base(pio);
        }
#endif
        pio->instr_mem[offset + i] = pio_instr_bits_jmp != _pio_major_instr_bits(instr) ? instr : instr + offset;
    }
    uint32_t program_mask = (1u << program->length) - 1;
    _used_instruction_space[pio_get_index(pio)] |= program_mask << offset;
    return (int)offset;
}
```

drewpo28, 2025-09-15:
> > Please remove the line - static_assert(PIO_GPIOBASE_BITS == 16, ""); // only works for gpio base being 0 or 16
> 
> Please clarify why you think this needs to be removed. I suspect you've jumped to the wrong conclusion about what it does.

I am using board Waveshare RP2350 PiZero.
It has following GPIO for HDMI

GPIO32 <-> DVI_D0_N
GPIO33 <-> DVI_D0_P
GPIO34 <-> DVI_D1_N
GPIO35 <-> DVI_D2_N
GPIO36 <-> DVI_D1_P
GPIO37 <-> DVI_D2_P
GPIO38 <-> DVI_CLK_P
GPIO39 <-> DVI_CLK_N

The example of code below is used to initialize hdmi and it not working without change proposed above.

`pio_set_gpio_base(PIO_VIDEO, 16);
pio_remove_program(PIO_VIDEO_ADDR, &pio_program_conv_addr_HDMI, offs_prg1);
offs_prg1 = pio_add_program(PIO_VIDEO_ADDR, &pio_program_conv_addr_HDMI);`

lurch, 2025-09-15:
> it not working without change proposed above.

That sounds very strange. As I pointed out above, `PIO_GPIOBASE_BITS` is a constant which always has the value 16 (i.e. 0x10 in hex) on RP2350, so `PIO_GPIOBASE_BITS == 16` should _always_ be true. Have you accidentally redefined `PIO_GPIOBASE_BITS` to some other value? Which board-header file are you using?

drewpo28, 2025-09-15:
I am using board pico2 with redefined #if PICO_PIO_USE_GPIO_BASE = 1
PIO_GPIOBASE_BITS is used by default

drewpo28, 2025-09-15:
I have tried it before without success, but will check again

kilograham, 2025-09-15:
if you are hitting that assert, you are probably building for RP2040 somehow. do you either specify PICO_PLATFORM=rp2350 when you build, or is it set via `pico_board_cmake_set(PICO_PLATFORM, rp2350)` in your board header?

Also, as mentioned before, you should not be setting `PICO_PIO_USE_GPIO_BASE = 1` - having to do so is likely a result of some other configuration issue. All in all, i'd say there are issues with your board header;