raspberrypi/pico-sdk #2675: could not convert 'main' from the host encoding (CP1252) to UTF-32.

StarflyerAlien, 2025-09-17:
First time starting a cortex-debug session with pico-sdk. Did not have this problem on the previous setup.
Session seems to run without problems. Debugging proceeding as expected with no issues.

Code is the standard blink example, unmodified!
Target: pico2_w
SDK :2.2.0

Message is:

> warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
> This normally should not happen, **please file a bug report**.

lurch, 2025-09-17:
Duplicate of https://github.com/raspberrypi/pico-vscode/issues/166 ? ping @will-v-pi

will-v-pi, 2025-09-17:
Yes, the error message is harmless - it seems to just be a bug in the arm-none-eabi-gdb debugger on Windows